raspberrypi/debugprobe #3: openocd with picoprobe segfaulting on windows

benevpi, 2021-01-27:
I followed the instructions in the Getting Started with Pico guide, and built OpenOCD for windows, but when I try to start it, it segfaults. I'm not even sure where to start with this. Is the info and warn details what are expected?

```
$ src/openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-27-11:20)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Segmentation fault
```

lurch, 2021-01-27:
paging @liamfraser

liamfraser, 2021-01-27:
I'll investigate this today

liamfraser, 2021-01-27:
This works with mingw-w64-x86_64-libusb-1.0.23-1-any.pkg.tar.xz. The latest libusb (mingw64/mingw-w64-x86_64-libusb 1.0.24-2) segfaults.

liamfraser, 2021-01-27:
I have built it here. Can you try it out?

https://drive.google.com/file/d/1SgZJepJWYQqCeC7m8se-yJLK9sv_AH8n/view?usp=sharing

benevpi, 2021-01-27:
I've only got one pico free at the moment. It outputs this:

```
C:\Users\ben\Downloads\openocd_picoprobe>openocd.exe  -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl            Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-27-15:43)                                                      Licensed under GNU GPL v2                                                                                               For bug reports, read                                                                                                           http://openocd.org/doc/doxygen/bugs.html                                                                        Info : only one transport option; autoselect 'swd'                                                                      Warn : Transport "swd" was already selected                                                                             adapter speed: 5000 kHz                                                                                                                                                                                                                         Info : Hardware thread awareness created                                                                                Info : Hardware thread awareness created                                                                                Info : RP2040 Flash Bank Command                                                                                        Info : Listening on port 6666 for tcl connections                                                                       Info : Listening on port 4444 for telnet connections                                                                    Info : clock speed 5000 kHz                                                                                             Info : DAP init failed  
```

Is this expected when it doesn't have another one to talk to? I'll try and test it out properly tomorrow.

liamfraser, 2021-01-27:
Yes. That looks like it's working

johnholman, 2021-01-28:
Just hit the same segfault problem with openocd so tried the build offered by @liamfraser. 
I got the same messages as above, but when I pulled out the USB connection to the pico Windows gave an error message about trying to reference memory at 0.

MimitechIndustries, 2021-02-03:
I encountered this as well, using the binary provided here for now.  Here is some basic stuff I got out of GDB:

```Thread 5 received signal SIGSEGV, Segmentation fault.
```

```Thread 5 (Thread 47944.0xbfa8):
#0  0x00007ff8c5d46e11 in ?? () from C:\msys64\mingw64\bin\libusb-1.0.dll
#1  0x00007ff8c5d49ffa in ?? () from C:\msys64\mingw64\bin\libusb-1.0.dll
#2  0x00007ff8f6e6b04a in msvcrt!_beginthreadex () from C:\WINDOWS\System32\msvcrt.dll
#3  0x00007ff8f6e6b11c in msvcrt!_endthreadex () from C:\WINDOWS\System32\msvcrt.dll
#4  0x00007ff8f7737bd4 in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#5  0x00007ff8f7d8ced1 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#6  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

MimitechIndustries, 2021-02-03:
I built both libusb v1.0.24 and the latest master manually and used the resulting DLLs -- v1.0.24 has the Segmentation Fault, latest master does not.  Here's a better backtrace for the thread that is segfault'ing:

```
Thread 5 (Thread 32732.0xbd4c):
#0  0x000000006b606e11 in usbi_signal_transfer_completion (itransfer=0x18df6d0) at io.c:1734
#1  0x000000006b609ffa in windows_iocp_thread (arg=0x3833900) at os/windows_common.c:442
#2  0x00007ff8f6e6b04a in msvcrt!_beginthreadex () from C:\WINDOWS\System32\msvcrt.dll
#3  0x00007ff8f6e6b11c in msvcrt!_endthreadex () from C:\WINDOWS\System32\msvcrt.dll
#4  0x00007ff8f7737bd4 in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#5  0x00007ff8f7d8ced1 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#6  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

planetchili, 2021-06-19:
Following the Getting Started guide and getting same segmentation fault issue.

maxgerhardt, 2021-07-15:
Just hit this problem too after compiling it myself using the official . 

When I run it with GDB I see 

```
[New Thread 19280.0x2f48]

Thread 5 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 19280.0x2f48]
0x00007ff807217734 in libusb_transfer_get_stream_id@4 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
(gdb) backtrace
#0  0x00007ff807217734 in libusb_transfer_get_stream_id@4 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
#1  0x00007ff80721a809 in libusb_interrupt_transfer@24 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
#2  0x00007ff866791bb2 in ucrtbase!_configthreadlocale () from C:\windows\System32\ucrtbase.dll
#3  0x00007ff867447034 in KERNEL32!BaseThreadInitThunk () from C:\windows\System32\kernel32.dll
#4  0x00007ff868da2651 in ntdll!RtlUserThreadStart () from C:\windows\SYSTEM32\ntdll.dll
#5  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

Mind that my `libusb-1.0.dll` comes from the [pyusb](https://github.com/pyusb/pyusb#installing) instructions telling me to use https://github.com/libusb/libusb/releases. 

Using the release provided above, it works nicely. It also works if I self-compile the latest branch and just put in the libusb-1.0.dll that has version info 1.0.23.11397. 

I think this is very crucial for Windows users and the instructions at https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf should be updated.

lurch, 2021-07-16:
> I think this is very crucial for Windows users and the instructions at https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf should be updated.

I think that requests for changes to the documentation should be created at https://github.com/raspberrypi/pico-feedback

maxgerhardt, 2021-07-17:
the linked libusb issue above I was also able to find out that when I use Zadig to load **WinUSB drivers** as opposed to libusb-win32 what the Raspberry Pi foundation's PDF is saying, using the latest released libusb-1.0.dll version 1.0.24 does **not** cause a crash. 

It is also explicitly said in https://github.com/libusb/libusb/issues/954#issuecomment-881788560 to **not** use use libusb-win32 but prefer WinUSB. 

Can someone reproduce this non-crashing on WinUSB with latest libusb-1.0.dll besides me?

mcuee, 2021-07-19:
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation
Recommended: Use the most recent version of Zadig, an Automated Driver Installer GUI application for WinUSB (recommended), libusb-win32 (not working well, not recommended) and libusbK (only if you hit WinUSB limitations)

mcuee, 2022-05-20:
The following libusb issue has been resolved in libusb release 1.0.25. Latest libusb version is 1.0.26.
* https://github.com/libusb/libusb/issues/954

But WinUSB driver is still recommended and not libusb0.sys.

SubaruArai, 2022-07-13:
Alright, I decided to bite the bullet and tried to cross-compile from Linux(Ubuntu 22.04 LTS) to Windows, and finally succeeded.

Also, I do highly recommend building in a container.

<details>
  <summary>build script</summary>

```bash
sudo apt install automake autoconf build-essential texinfo libtool pkg-config mingw-w64 autopoint flex cmake git

BUILD_DIR="$(pwd)/openocd_build"
LIBUSB1_BUILD_DIR=${BUILD_DIR}/libusb
LIBUSB0_BUILD_DIR=${BUILD_DIR}/libusb-compat
LIBCONFUSE_BUILD_DIR=${BUILD_DIR}/libconfuse
HIDAPI_BUILD_DIR=${BUILD_DIR}/hidapi
LIBFTDI_BUILD_DIR=${BUILD_DIR}/libftdi
mkdir -p ${BUILD_DIR}
pushd ${BUILD_DIR}

git clone --depth 1 --branch tags/v1.0.26 https://github.com/libusb/libusb.git libusb
pushd libusb
./bootstrap.sh
./configure --host=x86_64-w64-mingw32 --enable-static --disable-shared
make
popd

git clone --depth 1 --branch tags/v0.1.7 https://github.com/libusb/libusb-compat-0.1.git libusb-compat
pushd libusb-compat
./autogen.sh
LIBUSB_1_0_CFLAGS="-I${LIBUSB1_BUILD_DIR}/libusb" \
LIBUSB_1_0_LIBS="-L${LIBUSB1_BUILD_DIR}/libusb/.libs -lusb-1.0" \
PKG_CONFIG_PATH=${LIBUSB1_BUILD_DIR} \
./configure --host=x86_64-w64-mingw32 --enable-static --disable-shared
make
popd

git clone --depth 1 --branch tags/v3.3 https://github.com/martinh/libconfuse.git libconfuse
pushd libconfuse
./autogen.sh
# --disable-examples is needed for building with mingw
./configure --host=x86_64-w64-mingw32 --enable-static --disable-shared --disable-examples
make
popd

git clone --depth 1 --branch tags/v1.5 git://developer.intra2net.com/libftdi libftdi
pushd libftdi
mkdir build
pushd build
cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchain-x86_64-w64-mingw32.cmake \
 -DLIBUSB_LIBRARIES=${LIBUSB1_BUILD_DIR}/libusb/.libs/ \
 -DLIBUSB_INCLUDE_DIR=${LIBUSB1_BUILD_DIR}/libusb \
 -DCONFUSE_LIBRARY=${LIBCONFUSE_BUILD_DIR}/src/.libs/ \
 -DCONFUSE_INCLUDE_DIR=${LIBCONFUSE_BUILD_DIR}/src/ ../
make ftdi1-static
popd
popd

git clone --depth 1 --branch tags/hidapi-0.12.0 https://github.com/libusb/hidapi.git hidapi
pushd hidapi
./bootstrap
./configure --host=x86_64-w64-mingw32 --enable-static --disable-shared
make
popd

git clone https://github.com/raspberrypi/openocd.git --branch rp2040
pushd openocd
# this is the tested version of openocd, change as needed
git checkout 228ede43db3665e470d2e518730de013a8c74411
./bootstrap
HIDAPI_CFLAGS="-I${HIDAPI_BUILD_DIR}/hidapi" \
HIDAPI_LIBS="-L${HIDAPI_BUILD_DIR}/windows/.libs -lhidapi" \
LIBUSB0_CFLAGS="-I${LIBUSB0_BUILD_DIR}/libusb" \
LIBUSB0_LIBS="-L${LIBUSB0_BUILD_DIR}/libusb/.libs -lusb" \
LIBUSB1_CFLAGS="-I${LIBUSB1_BUILD_DIR}/libusb" \
LIBUSB1_LIBS="-L${LIBUSB1_BUILD_DIR}/libusb/.libs -lusb-1.0" \
LIBFTDI_CFLAGS="-I${LIBFTDI_BUILD_DIR}/src " \
LIBFTDI_LIBS="-L${LIBFTDI_BUILD_DIR}/build/src -lftdi1" \
PKG_CONFIG_PATH=${HIDAPI_BUILD_DIR}/pc:${LIBUSB1_BUILD_DIR}:${LIBUSB0_BUILD_DIR}:${LIBFTDI_BUILD_DIR}/build \
libusb_CFLAGS="-I${LIBUSB1_BUILD_DIR}/libusb" \
libusb_LIBS="-L${LIBUSB1_BUILD_DIR}/libusb/.libs/ -lusb-1.0" \
./configure --host=x86_64-w64-mingw32 --enable-static --disable-shared --enable-picoprobe
make
popd

# now copy the built files to your windows system with the following structure:
# openocd
# |-openocd.exe (copy from openocd/src/openocd.exe)
# |-tcl (copy from openocd/tcl)
# | |- (everything under tcl must be here)

# `.\openocd.exe -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl` to run in windows
```
</details>

Credits goes to [this blog post](https://qiita.com/qawsed477/items/2f93f2c1ff0de6fef038) and openocd's [readme](https://openocd.org/doc-release/README).

mcuee, 2022-07-13:
It is actually pretty easy to build openocd under Windows natively using MSYS2 ming32/mingw64. They have the formula for latest version of libusb-1.0, libusb-compat-0.1, libftdi1 and hidapi. They also have capstone. So you do not need to build them by yourselves.

SubaruArai, 2022-07-14:
@mcuee My bad, didn't notice that.
I recently tried to build under msys2 but failed, which led to me doing cross-compile.

I also find it being much more stable - everything is statically linked, and every library version (except mingw) is defined (no more build failure due to system updates!).
Btw, being in a container is a HUGE benefit, IMHO.

But then again, this is going off-topic, so I'd end my bragging.

mcuee, 2022-07-14:
All in all, I think this issue can be closed as the upstream libusb issue has been resolved. You should really use libusb-1.0.26 for new build of OpenOCD.

For people who are still using libusb-1.0.24, you can have a easy work-around as well -- just use Zadig to install WinUSB driver.

Gnomey123, 2022-09-01:
> ```shell
> git clone --depth 1 --branch tags/v0.1.7 https://github.com/libusb/libusb-compat-0.1.git libusb-compat
> ```

I'd like to update this script since the git clone branches aren't correct.

Modifying them fixed it for me.

[script.txt](https://github.com/raspberrypi/picoprobe/files/9471553/script.txt)

lurch, 2022-09-01:
> I'd like to update this script since the git clone branches aren't correct.

Huh, which script are you referring to exactly?

Gnomey123, 2022-09-01:
oops, forgot to properly attach quote by [SubaruArai](https://github.com/SubaruArai)

closed 2023-02-15 by P33M