raspberrypi/debugprobe #5: No license is no license

newbrain, 2021-02-01:
Thanks for making the code available, but no License (or a readme with a license description) is provided in root and while many source files do have an embedded MIT (and in one case a BSD-3-Clause) license statement, others are completely unlicensed.

As I am sure you know, no license == no license, so hindering the redistribution and modification of the code, if one wants to follow the rules.

Missing licenses:
* src/cdc_uart.c 
* src/cdc_uart.h
* src/picoprobe_config.h
* src/probe.h
* src/probe.pio

newbrain, 2021-02-01:
BTW: I noticed this as I wanted to add a way to generate an unique serial number for each probe.
As the RP2040 does not have any unique ID register (the CHIP_ID suggested in the code is as good as the current "123456", I think the only way is to make some XIP contortions and read the serial flash identifier with command 0x4B (which should be standard).
I'll try to code this in the coming days.

kilograham, 2021-02-01:
a method for this (getting a unique serial number from the flash) was just added to the SDK release 1.0.1 now out

liamfraser, 2021-02-03:
I have resolved this. Thanks

closed 2021-02-03 by liamfraser