raspberrypi/debugprobe #12: error: expected initializer before 'noexcept'  void operator delete(void *p, std::size_t n) noexcept { std::free(p); }

tge96, 2021-02-24:
Hi,

[100%] Linking CXX executable pioasm
[100%] Built target pioasm
[ 20%] No install step for 'PioasmBuild'
[ 21%] Completed 'PioasmBuild'
[ 21%] Built target PioasmBuild
Scanning dependencies of target picoprobe_probe_pio_h
[ 22%] Generating probe.pio.h
[ 22%] Built target picoprobe_probe_pio_h
Scanning dependencies of target picoprobe
[ 23%] Building C object CMakeFiles/picoprobe.dir/src/led.c.obj
[ 24%] Building C object CMakeFiles/picoprobe.dir/src/main.c.obj
[ 25%] Building C object CMakeFiles/picoprobe.dir/src/usb_descriptors.c.obj
[ 26%] Building C object CMakeFiles/picoprobe.dir/src/probe.c.obj
[ 27%] Building C object CMakeFiles/picoprobe.dir/src/cdc_uart.c.obj
[ 29%] Building C object CMakeFiles/picoprobe.dir/src/get_serial.c.obj
[ 30%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[ 31%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[ 32%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[ 33%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
[ 34%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[ 35%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_uart/uart.c.obj
[ 36%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_divider/divider.S.obj
[ 37%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_time/time.c.obj
[ 38%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_time/timeout_helper.c.obj
[ 39%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_timer/timer.c.obj
[ 40%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_sync/sem.c.obj
[ 41%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_sync/lock_core.c.obj
[ 43%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_sync/mutex.c.obj
[ 44%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_sync/critical_section.c.obj
[ 45%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_util/datetime.c.obj
[ 46%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_util/pheap.c.obj
[ 47%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/common/pico_util/queue.c.obj
[ 48%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj
[ 49%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_clocks/clocks.c.obj
[ 50%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_watchdog/watchdog.c.obj
[ 51%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_xosc/xosc.c.obj
[ 52%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_pll/pll.c.obj
[ 53%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_vreg/vreg.c.obj
[ 54%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[ 55%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
[ 56%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
[ 58%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj
[ 59%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c.obj
[ 60%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_divider/divider.S.obj
[ 61%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_double/double_aeabi.S.obj
[ 62%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_double/double_init_rom.c.obj
[ 63%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
[ 64%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[ 65%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[ 66%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
[ 68%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
[ 68%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
[ 69%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
[ 70%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
[ 72%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
[ 73%] Building ASM object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
[ 74%] Building CXX object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:20:46: error: expected initializer before 'noexcept'
 void operator delete(void *p, std::size_t n) noexcept { std::free(p); }
                                              ^
/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:24:33: error: expected initializer before 'noexcept'
 void operator delete[](void *p) noexcept { std::free(p); }
                                 ^
CMakeFiles/picoprobe.dir/build.make:585: recipe for target 'CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj' failed
make[2]: *** [CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:74: recipe for target 'CMakeFiles/picoprobe.dir/all' failed
make[1]: *** [CMakeFiles/picoprobe.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~ $

kilograham, 2021-02-24:
try adding  

set(CMAKE_CXX_STANDARD 11)

to your CMakeLists.txt

I'm guessing you have a back-level compiler though.

tge96, 2021-02-24:
Hi, this error message is coming when running pico_setup.sh

tge96, 2021-02-24:
Is pico_setup.sh suitable for Raspberry Pi 3B+ and Raspberry Pi Zero W?  Or does it require Raspberry Pi 4B+?

lurch, 2021-02-24:
I believe as far as `pico_setup.sh` is concerned, there should be no difference between _any_ of the Raspberry Pi Models, since they all run the same Raspberry Pi OS, using the same APT repositories.

What OS are you running on your Raspberry Pi 3B+ ? What does `gcc --version` and `arm-none-eabi-gcc --version` say?

tge96, 2021-02-24:
pi@rpi3bp:~ $ gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@rpi3bp:~ $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

aallan, 2021-02-24:
Both of these are massively out of date, you're not running Stretch are you? Those are ancient. This from the Pi on my desk. 

```
pi@raspberrypi:~ $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@raspberrypi:~ $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@raspberrypi:~ $ 
```

If so you should just go ahead and do a fresh OS install,  you need an up to date version of the operating system.

tge96, 2021-02-24:
yes, I just discovered in another post that my Raspberry Pi3B+ is running stretch.  is it necessary to do a clean install?  or can I upgrade stretch to buster?

closed 2021-02-25 by tge96