raspberrypi/debugprobe #15: Don't use Zadig for Windows driver installs

Gadgetoid, 2021-05-24:
Zadig is a slow, prone to error and fairly overcomplicated to install a driver for Picoprobe on Windows. You should use LibusK-inf-wizard instead and vendor a driver into a Windows installer package.

Attached example targets the Picoprobe VID/PID (Interface 2) and installs in one click (clicking Finish and, uh, extracting the installer notwithstanding).

Windows users might be familiar with this screen from the driver installs of other micro manufacturers:

[image]

Yes you have to re-vender if the libUSB version you want to target changes... but if that's a breaking change (do these happen with libUSB windows drivers? I don't know) then right now the first you'll hear of it is bug reports.

[Picoprobe_libUSB.zip](https://github.com/raspberrypi/picoprobe/files/6533810/Picoprobe_libUSB.zip)

lurch, 2021-05-24:
https://sourceforge.net/projects/libusbk/files/ says the last update was in 2014? :confused:

Gadgetoid, 2021-05-24:
You don’t need to mess with perfection 🤣

lurch, 2021-05-24:
I'm not sure if there's enough spare engineer-time for anyone to change our existing approach to USB support on Windows? :man_shrugging:  ping @aallan

kilograham, 2021-05-24:
If this is "easier" than zadig; i'll all for it; @Gadgetoid ... can you explain the steps at a high level so we don't have to go figure it out!

mcuee, 2021-07-19:
Pull request #17 is the best method for Windows 8.1/Windows 10. Before that happens, you can create a driver installation package if you want to make it very user friendly. Both Zadig and libusbk provide that option.

As for libusbk-inf-wizard, we just released 3.0.8 version. As for Zadig vs libusbk-inf-wizard, it is up to the individual. Both are based on libwdi.

libusbk release 3.0.8.0:
https://github.com/mcuee/libusbk/releases
https://sourceforge.net/projects/libusbk/files/libusbK-release/3.0.8.0/

mcuee, 2021-07-19:
Main issue is not Zadig vs libusbk-inf-wizard, but rather please change the documentation to use WinUSB driver and not libusb-win32 device driver in the documentation.

Ref: https://github.com/raspberrypi/picoprobe/issues/3 and https://github.com/libusb/libusb/issues/954

mcuee, 2021-07-19:
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation
Recommended: Use the most recent version of Zadig, an Automated Driver Installer GUI application for WinUSB (recommended), libusb-win32 (not working well, not recommended) and libusbK (only if you hit WinUSB limitations)

sdbbs, 2022-05-20:
Hi all,

Just wanted to note my unsuccesful results in Windows 10, following some of the recommendations here.

First of all, I built the "vanilla" picoprobe (under MSYS2's MINGW64), flashed the .uf2 on my Pico, connected the Pico to Windows 10 PC. Windows Device Manager said:

* Other devices: Picoprobe with yellow exclamation mark ("The drivers for this device are not installed.")
* and Ports (COM & LPT): USB Serial Device (COM12)

Then I found this thread, and I thought I'd give Pull request https://github.com/raspberrypi/picoprobe/pull/17 a try:

```
git clone https://github.com/ciniml/picoprobe.git --branch wcid picoprobe_ciniml_wcid
cd picoprobe_ciniml_wcid
mkdir build
cd build
cmake .. -G "MSYS Makefiles"
make -j4
```

Flashed this .uf2 on the Pico - with that, Windows Device Manager said:

* Ports (COM & LPT)
    * USB Serial Device (COM12)
* Universal Serial Bus Devices
    * Picoprobe (no exclamation mark! "This device is working properly." Driver provider: libwdi, Digital Signer: USB\MS_COMP_WINUSB (libwdi autogenerated))

I'd like to note here, that https://visualgdb.com/tutorials/raspberry/pico/picoprobe/ mentions:

> Raspberry Pi will restart and appear in the Device Manager as a special Picoprobe device paired with a virtual COM port:

... but that is not at all how it appeared in my Device Manager at this point: I did not have a single node of paired devices - rather, I had two separate nodes, as noted above.

At this point, I thought about inspecting with devcon.exe as well - considering the USB VID:PID of Picoprobe is 2E8A:0004 : 

```
$ /c/bin/devcon.exe driverfiles '*2E8A*'
USB\VID_2E8A&PID_0004&MI_02\8&28DF7DA2&0&0002
    Name: Picoprobe
    Driver installed from C:\WINDOWS\INF\oem63.inf [USB_Install]. 3 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\winusb.sys
        C:\WINDOWS\system32\WinUSBCoInstaller2.dll
        C:\WINDOWS\system32\WdfCoInstaller01011.dll
USB\VID_2E8A&PID_0004\E660D4A0A75B7834
    Name: USB Composite Device
    Driver installed from C:\WINDOWS\INF\usb.inf [Composite.Dev]. 1 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\usbccgp.sys
USB\VID_2E8A&PID_0004&MI_00\8&28DF7DA2&0&0000
    Name: USB Serial Device (COM12)
    Driver installed from C:\WINDOWS\INF\usbser.inf [UsbSerial_Install]. 1 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\usbser.sys
3 matching device(s) found.
```

At this point, I get openocd error:

```
$ ./src/openocd.exe -s ./tcl -f interface/picoprobe.cfg -f target/rp2040.cfg -c "init; reset; exit"
Open On-Chip Debugger 0.11.0-g610f137 (2022-03-29-13:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

Noting the advice in https://github.com/raspberrypi/picoprobe/issues/15#issuecomment-882426440 :

> https://github.com/libusb/libusb/wiki/Windows#Driver_Installation Recommended: Use the most recent version of Zadig, an Automated Driver Installer GUI application for WinUSB (recommended), libusb-win32 (not working well, not recommended) and libusbK (only if you hit WinUSB limitations)

... I fired up Zadig-2.7; it saw:

* Picoprobe (Interface 0) - driver usbser (v10.0.19041.1202)
* Picoprobe (Interface 2) - driver WinUSB (v6.1.7600.16385)

So, since  (Interface 2) already had driver WinUSB (v6.1.7600.16385) - I decided to only replace the driver for Picoprobe (Interface 0) - from driver usbser (v10.0.19041.1202), to driver WinUSB (v6.1.7600.16385).

Now, after USB replug of the picoprobe Pico: I do NOT have USB Serial Device (COM12) under Ports (COM & LPT) anymore -- I only have two Picoprobe entries, under:

* Universal Serial Bus Devices
    * Picoprobe (no exclamation mark)
    * Picoprobe (no exclamation mark)

At this time, devcon.exe sees:

```
$ /c/bin/devcon.exe driverfiles '*2E8A*'
USB\VID_2E8A&PID_0004&MI_02\8&28DF7DA2&0&0002
    Name: Picoprobe
    Driver installed from C:\WINDOWS\INF\oem63.inf [USB_Install]. 3 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\winusb.sys
        C:\WINDOWS\system32\WinUSBCoInstaller2.dll
        C:\WINDOWS\system32\WdfCoInstaller01011.dll
USB\VID_2E8A&PID_0004\E660D4A0A75B7834
    Name: USB Composite Device
    Driver installed from C:\WINDOWS\INF\usb.inf [Composite.Dev]. 1 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\usbccgp.sys
USB\VID_2E8A&PID_0004&MI_00\8&28DF7DA2&0&0000
    Name: Picoprobe
    Driver installed from C:\WINDOWS\INF\oem65.inf [USB_Install]. 3 file(s) used by driver:
        C:\WINDOWS\system32\DRIVERS\winusb.sys
        C:\WINDOWS\system32\WinUSBCoInstaller2.dll
        C:\WINDOWS\system32\WdfCoInstaller01011.dll
3 matching device(s) found.
```

... and openocd now fails with a different message:

```
$ ./src/openocd.exe -s ./tcl -f interface/picoprobe.cfg -f target/rp2040.cfg -c "init; reset; exit"
Open On-Chip Debugger 0.11.0-g610f137 (2022-03-29-13:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Error: Failed to claim picoprobe interface
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

Could anyone maybe document, how should a working device driver setup of a Picoprobe Pico on Windows 10 look like, with [DevCon.exe](https://superuser.com/questions/1002950/quick-method-to-install-devcon-exe)?

mcuee, 2022-05-20:
Your driver installaiton is already correct with #17 (with COM12 and then another device Picoprobe with WinUSB driver). You do not need to use Zadig again.

Your issue with OpenOCD is a different issue. You may need to build your OpenOCD binary again.

mcuee, 2022-05-20:
The following libusb issue has been resolved in libusb release 1.0.25. Latest libusb version is 1.0.26.
* https://github.com/libusb/libusb/issues/954
* 
But WinUSB driver is still recommended and not libusb0.sys.

mcuee, 2022-05-20:
@sdbbs You may want to create a new issue on OpenOCD. You need to mention how you build it and the debug log.

Did you use the following? If yes you want to create a issue there.
https://github.com/raspberrypi/openocd/

sdbbs, 2022-05-20:
Thanks, @mcuee : 

> Did you use the following? 
>
> https://github.com/raspberrypi/openocd/
>
> If yes you want to create a issue there.

Indeed, I am - I will try to poke a bit more, and if I do not get it running soon, I'll probably post an issue there.

Thanks especially for the confirmation:

> Your driver installaiton is already correct with https://github.com/raspberrypi/picoprobe/pull/17 (with COM12 and then another device Picoprobe with WinUSB driver). You do not need to use Zadig again.

... so at least I know what is the right driver state.

Btw, to recover from the last step I made in previous post: find the Picoprobe node, which upon right-click in Device Manager / Details says Value: "Picoprobe (Interface 0)" under Property: Device Description in the Details tab; then in Driver tab, hit Uninstall Device, and on the prompt, also check "Delete the driver software for this device.", and hit "Uninstall". The node will disappear from the Device Manager tree; then replug the picoprobe Pico via USB - and one should be back to one Picoprobe device under Universal Serial Bus devices, and one USB Serial Device under Ports (COM & LPT).

mcuee, 2022-05-20:
BTW, you actually do not need #17, that just makes things easier.
Even without #17, you can use Zadig to install WinUSB driver and then the yellow mark in Device Manager will disappear.

mcuee, 2022-07-14:
I think this issue can be closed. 
Either Zadig or libusbk-inf-wizard can be used.

Gadgetoid, 2022-07-14:
@mcuee I don't think a standalone, beginner-friendly driver has been packaged yet? But if Zadig is the recommended approach- and it's worked so far- then it seems perhaps a little redundant to mess with it now.

P33M, 2022-11-11:
Fixed in #31 - Zadig is no longer required for both interfaces to work.

closed 2022-11-11 by P33M