raspberrypi/debugprobe #16: Release flashable .uf2 files

maxgerhardt, 2021-07-05:
For people to easily flash their devices with the Picoprobe firmware, this repo should provide precomompiled .uf2 files that people can just drag-and-drop onto a Pico in bootloader mode.

aallan, 2021-07-05:
> For people to easily flash their devices with the Picoprobe firmware, this repo should provide precomompiled .uf2 files that people can just drag-and-drop onto a Pico in bootloader mode, either as a file in the repo or as a Github release.

There is a precompiled UF2 available on the Pico Getting Started pages, see https://www.raspberrypi.org/documentation/rp2040/getting-started/#board-specifications, and scroll to the "Utilities" section.

closed 2021-07-05 by aallan

maxgerhardt, 2021-07-05:
Why is that on a different website, https://www.raspberrypi.org/documentation/rp2040/getting-started/static/fec949af3d02572823529a1b8c1140a7/picoprobe.uf2 ? :D  

I think at least a small hint in the `README.md` would still be good.

aallan, 2021-07-05:
That is the website you end up on when you click through from INDEX.HTM file on your Pico when it's mounted as a mass storage device. The website and the UF2 are discussed in the Getting Started documentation which talks about Picoprobe. This repo isn't documented here (there is no README.md file at all). Pico documentation lives elsewhere and points at this repo.

maxgerhardt, 2021-07-05:
I see.