raspberrypi/debugprobe #22: UART doesn't seem to work

Pyjong, 2022-03-26:
Hello,

I've flashed picoprobe fw to my pico via dropping uf2 file onto presented mass storage. I've noticed `picoprobe_info()` is disabled by default so I changed that false to true in picoprobe_config.h in expectation to see the "Welcome to picoprobe" message in PuTTY (bitrate 115200). That didn't really happen though, so I changed it for printf, but still no message. 
I also wanted to make a little sanity check so I added some LED blinking and that worked.
Would you have an idea what might be going wrong?

lurch, 2022-03-26:
Are you connecting PuTTY to the USB-CDC UART? It's possible that the "Welcome to picoprobe" has already been "printed" before the USB serial port enumerates on your PC? :shrug:

Pyjong, 2022-03-26:
True, but I moved the message down to the main loop where I added the led blinking so it should get printed periodicaly.
This is how my main looks now:
`int main(void) {
    int i = 0;
    unsigned char led = 0;
    board_init();
    usb_serial_init();
    cdc_uart_init();
    tusb_init();
    probe_init();
    led_init();

    picoprobe_info("Welcome to Picoprobe!\n");

    while (1) {
        tud_task(); // tinyusb device task
        cdc_task();
        probe_task();
        led_task();
        if (i++>=500000)
        {
            gpio_put(PICOPROBE_LED, led);
            printf("Welcome to Picoprobe!\n");
            i = 0;
            led ^= 1;
        }
    }

    return 0;
}`

it turns the LED on/off about every second which seems slow, but of course the tasks have some to me undefined overhead too.

Pyjong, 2022-03-26:
I think I'm connecting PuTTY to the correct serial device - it disappears when I unplug the cable and reappears when I plug it back in.

lurch, 2022-03-26:
ping @liamfraser

liamfraser, 2022-03-28:
The picoprobe debug info comes out of UART0, which is on pins 0 and 1 of the board. Is that what you are connected to?

Pyjong, 2022-03-29:
Ah yes. I had it right on the pico but it was wrong on my support board that has the ftdi x) Thanks.

lurch, 2022-03-29:
Can this be closed @Pyjong ?

liamfraser, 2022-03-29:
Looks like it. Closing.

closed 2022-03-29 by liamfraser

Pyjong, 2022-03-29:
Right, yes, please close. Thank you!

Pyjong, 2022-03-29:
Erm me again but this time I think I might have something useful. So first thanks for helping me, with UART I could debug the other issue I had with SWD clock. So picoprobe code is correct. The clock was actually set incorrectly though. There is a minor quirk when it comes to cross-compilation with MinGW for Windows - `__attribute__(packed)` doesn't work unless you specify `-mno-ms-bitfields`. Not sure if this switch finds place in somewhere openocd. Alternatively:
```
struct __attribute__((__packed__)) probe_cmd_hdr {
	uint32_t bits;
    uint8_t id;
	uint8_t cmd;
};
```
fixes the issue if done on both picoprobe fw and openocd driver side. I hope this helps.

Vitch612, 2023-02-05:
> 
arm-none-eabi-gcc for windows does not accept the -mno-ms-bitfields option. Does anyone know a workaround for this? Or could it be that the option is not recognized because it is enabled by default on the windows version of arm-none-eabi-gcc?
I am trying this with GNU Arm Embedded Toolchain v10 2021.10 on windows 7 64bit

aallan, 2023-02-05:
> …on windows 7 64bit

We really don't support RP2040 development on version of Windows before Windows 10. Extended support for Windows 7 ended on January 14, 2020?

Vitch612, 2023-02-05:
> > …on windows 7 64bit
> 
> We really don't support RP2040 development on version of Windows before Windows 10. Extended support for Windows 7 ended on January 14, 2020? That was over a year _before_ the Raspberry Pi Pico was released.

I know that, I mentionned the version to give as much info as I can, the question is not really OS version dependant the GNU Arm toolchain does not have different versions for each version of windows and I doubt that setting -mno-ms-bitfields on windows 11 would work either.

dp111, 2023-02-05:
Can you check which version of gcc is being used : arm-none-eabi-gcc -v

lurch, 2023-02-06:
@Pyjong mentioned that he was using "MinGW for Windows", so perhaps that has different compiler-flags to "GNU Arm Embedded Toolchain" being used by @Vitch612 ? :shrug:

Vitch612, 2023-02-06:
Hello guys many thanks for reacting so quickly and trying to help.

For info it turns out that neither GNU Arm Embedded Toolchain v10 2021.10 nor MSYS2 gcc (both latest at the time of the this post) accept the flag -mno-ms-bitfields. 
version `(Rev10, Built by MSYS2 project) 12.2.0` for MSYS2  MINGW64 gcc.
and version  `(GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)` for GNU Arm Embedded Toolchain. 

However this was not needed (I am guessing that being both meant to run on windows this flag is implied and not needed as an option). It turned out that everything is working fine both in picoprobe mode and in PROTO_DAP_V2 mode. (PROTO_DAP_V2 tried from compiling a slightly modified picoprobe pull (to work with pico_w) while picoprobe version was tried from downloading a precompiled version from [here](https://github.com/Fabien-Chouteau/picoprobe-pcb#install-the-picoprobe-software)

The error was in my debugged device code where i was reseting all GPIO ports to GPIO_FUNC_NULL after the stdio_init_all() call.

Conclusion is that it is all fully functional and works well on windows 7x64 with VSCode 1.70.3 (last win7 compatible version)

lurch, 2023-02-06:
> tried from compiling a slightly modified picoprobe pull (to work with pico_w) while picoprobe version was tried from downloading a precompiled version

AIUI, the `picoprobe.uf2` linked to from https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#debugging-using-another-raspberry-pi-pico supports both Pico and Pico W, and both the "older" picoprobe interface and the "newer" CMSIS DAP interface?