raspberrypi/debugprobe #23: Open OCD with Picoprobe

BabumonGopi, 2022-04-08:
When I try to connect Picoprobe for debugging (Installed all the softwares in my Mac as per instructions). Also connected as per the schematic given. But the Open OCD is giving the following error.
"Error: The specified debug interface was not found (picoprobe)".
I think the Picoprobe is not detecting by the OpenOCD.
But when I check the USB Device tree, the Picoprobe shows in it there when connecting.

Please help what to do to correct it.

lurch, 2022-04-08:
As per the instructions, have you also compiled the custom version of openocd that's needed? See Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

BabumonGopi, 2022-04-08:
Sir,
       Thank you for your response. Sure, I had compiled the open ocd as per the Appendix A. Please see below the response from OpenOcd.

Last login: Fri Apr  8 10:35:44 on ttys000
***@***.*** ~ % cd pico
***@***.*** pico % cd openocd
***@***.*** openocd % cd tcl
***@***.*** tcl % openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (picoprobe)
The following debug adapters are available:
1: dummy
2: ftdi
3: usb_blaster
4: jtag_vpi
5: ft232r
6: presto
7: usbprog
8: openjtag
9: jlink
10: vsllink
11: rlink
12: ulink
13: arm-jtag-ew
14: buspirate
15: remote_bitbang
16: hla
17: osbdm
18: opendous
19: aice
20: cmsis-dap
21: kitprog
22: xds110
23: st-link
***@***.*** tcl % 

Please help to solve this issue.

lurch, 2022-04-08:
The instructions show that you need to run `src/openocd` in order to run the custom-compiled openocd. You're running just `openocd` which probably means that it's running the "system-installed" version of openocd instead?

BabumonGopi, 2022-04-08:
Sir,
        Can’t make it. Error occurs please see below.

***@***.*** openocd % make -j4
cat src/helper/startup.tcl src/jtag/startup.tcl src/target/startup.tcl src/server/startup.tcl src/flash/startup.tcl | ./src/helper/bin2char.sh > src/startup_tcl.inc || { rm -f src/startup_tcl.inc; false; }
cp src/jtag/drivers/minidriver_imp.h src/jtag/minidriver_imp.h
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in jimtcl
	MKLDEXT	_load-static-exts.c
	CC	jim-subcmd.o
	CC	jim-interactive.o
	CC	jim-format.o
	CC	jim.o
	CC	utf8.o
	CC	jimregexp.o
	CC	jimiocompat.o
	CC	linenoise.o
	CC	jim-tty.o
	CC	jim-aio.o
	CC	jim-array.o
	CC	jim-clock.o
	CC	jim-eventloop.o
	CC	jim-exec.o
	CC	jim-file.o
jim-exec.c:1046:15: warning: 'vfork' is deprecated: Use posix_spawn or fork [-Wdeprecated-declarations]
        pid = vfork();
              ^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.0.sdk/usr/include/unistd.h:604:1: note: 'vfork' has been explicitly marked deprecated here
__deprecated_msg("Use posix_spawn or fork")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.0.sdk/usr/include/sys/cdefs.h:208:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
jim-exec.c:729:12: warning: variable 'child_environ' set but not used [-Wunused-but-set-variable]
    char **child_environ;
           ^
	CC	jim-history.o
	CC	jim-interp.o
2 warnings generated.
	CC	jim-load.o
	CC	jim-namespace.o
	CC	jim-pack.o
	CC	jim-package.o
	CC	jim-posix.o
	CC	jim-readdir.o
	CC	jim-regexp.o
	CC	jim-signal.o
	CC	jim-syslog.o
 	TCLEXT	_glob.c
 	TCLEXT	_nshelper.c
 	TCLEXT	_oo.c
 	CC	glob.o
 	CC	nshelper.o
 	CC	oo.o
 	TCLEXT	_stdlib.c
 	TCLEXT	_tclcompat.c
 	TCLEXT	_tree.c
 	CC	stdlib.o
 	CC	tree.o
 	CC	tclcompat.o
	CC	jimsh.o
 	TCLEXT	_initjimsh.c
 	CC	initjimsh.o
	CC	_load-static-exts.o
	AR	libjim.a
	LINK	jimsh
Making all in src/jtag/drivers/libjaylink
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libjaylink
/bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden -I/opt/homebrew/Cellar/libusb/1.0.25/include/libusb-1.0 -g -O2 -MT libjaylink_la-buffer.lo -MD -MP -MF .deps/libjaylink_la-buffer.Tpo -c -o libjaylink_la-buffer.lo `test -f 'buffer.c' || echo './'`buffer.c
[... 109 lines cut ...]
	mv -f $depbase.Tpo $depbase.Po
/bin/sh ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -DRELSTR=\"`./guess-rev.sh .`\" -DGITVERSION=\"`cd . && git describe`\" -DPKGBLDDATE=\"`date +%F-%R`\"  -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl   -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -g -O2 -MT src/libopenocd_la-hello.lo -MD -MP -MF src/.deps/libopenocd_la-hello.Tpo -c -o src/libopenocd_la-hello.lo `test -f 'src/hello.c' || echo './'`src/hello.c
./doc/openocd.texi:10195: warning: `.' or `,' must follow @xref, not `c'.
./doc/openocd.texi:10934: Unknown command `raggedright'.
./doc/openocd.texi:10939: Bad argument `raggedright' to ***@***.***', using `table'.
./doc/openocd.texi:10939: @item found outside of an insertion block.
./doc/openocd.texi:10941: @item found outside of an insertion block.
./doc/openocd.texi:10943: @item found outside of an insertion block.
./doc/openocd.texi:10946: @item found outside of an insertion block.
./doc/openocd.texi:10948: @item found outside of an insertion block.
./doc/openocd.texi:10950: @item found outside of an insertion block.
./doc/openocd.texi:10952: @item found outside of an insertion block.
./doc/openocd.texi:10953: Unknown command `raggedright'.
./doc/openocd.texi:10957: Unmatched ***@***.***'.
./doc/openocd.texi:10958: Unmatched ***@***.***'.
makeinfo: Removing output file `doc/openocd.info' due to errors; use --force to preserve.
make[2]: *** [doc/openocd.info] Error 1
make[2]: *** Waiting for unfinished jobs....
mv -f src/jtag/drivers/.deps/libocdjtagdrivers_la-xds110.Tpo src/jtag/drivers/.deps/libocdjtagdrivers_la-xds110.Plo
fatal: No names found, cannot describe anything.
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -DRELSTR=\"-g610f137\" -DGITVERSION=\"\" -DPKGBLDDATE=\"2022-04-08-23:18\" -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -g -O2 -MT src/libopenocd_la-hello.lo -MD -MP -MF src/.deps/libopenocd_la-hello.Tpo -c src/hello.c -o src/libopenocd_la-hello.o
mv -f src/.deps/libopenocd_la-hello.Tpo src/.deps/libopenocd_la-hello.Plo
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
***@***.*** openocd %

lurch, 2022-04-11:
It's failing while trying to build the openocd documentation. Are you sure you've followed the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf carefully? It says "The version of texinfo shipped with OSX is below the version required to build OpenOCD docs".

BabumonGopi, 2022-04-11:
Sir,
      Now another error. 

src/target/arm_disassembler.c:30:10: fatal error: 'capstone/capstone.h' file not found
#include <capstone/capstone.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.

> On 11-Apr-2022, at 5:08 PM, Andrew Scheller ***@***.***> wrote:
> 
> The version of texinfo shipped with OSX is below

aallan, 2022-04-11:
This feels like the sort of support question you should be asking on the [forums](https://www.raspberrypi.org/forums/). It's not an issue or an error directly with Picoprobe, so I’m closing this issue.

closed 2022-04-11 by aallan

BabumonGopi, 2022-04-11:
> fatal error: 'capstone/capstone.h' file not found
>
> See raspberrypi/openocd#30 <https://github.com/raspberrypi/openocd/issues/30> and raspberrypi/openocd#7 <https://github.com/raspberrypi/openocd/issues/7>

Thank you for your support. It is working fine now. I had changed the include path of capstone.h in two files (openocd/src/target/a64_disassembler.c, openocd/src/target/arm_disassembler.c) as shown below.

#include <capstone/capstone.h>

Changed to

#include "/opt/homebrew/Cellar/capstone/4.0.2/include/capstone/capstone.h”

BabumonGopi, 2022-04-21:
Dear Sir,
                Any working program for Mass Storage Device working with pico-sdk ? I had tried to connect with TinyUSB, but it will not working for pico. The following error is coming.

assertion "ep" failed: file "/Users/babumongopi/r-pi/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 492, function: hcd_edpt_xfer

Any solution ?

Babumon Gopi

Poornasree Equipments
www.poornasree.com

lurch, 2022-04-21:
That's totally unrelated to your original question, so you should post a new issue rather than commenting on an already-closed one.