raspberrypi/debugprobe #24: No SWD Activity from Picoprobe

dchokola, 2022-04-08:
1. programmed 2 separate Picos with both the prebuilt picoprobe.uf2 and one I compiled myself
2. hooked up a logic analyzer to the SWDIO and SWDCK pins
3. verified the Picoprobe is seen by the OS:
```
[1329357.426992] usb 4-2: new full-speed USB device number 124 using uhci_hcd
[1329357.615140] usb 4-2: New USB device found, idVendor=2e8a, idProduct=0004, bcdDevice= 1.00
[1329357.615150] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1329357.615154] usb 4-2: Product: Picoprobe
[1329357.615156] usb 4-2: Manufacturer: Raspberry Pi
[1329357.615158] usb 4-2: SerialNumber: E6616407E3663127
[1329357.620283] cdc_acm 4-2:1.0: ttyACM0: USB ACM device
^C
```
4. ran openocd (full debug output attached)
[openocd-debug.txt](https://github.com/raspberrypi/picoprobe/files/8455666/openocd-debug.txt)

```
dan@ender:~/pico/openocd$ openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g610f137 (2022-04-08-16:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Info : DAP init failed

```
5. logic analyzer shows both SWD pins held high the entire time
[openocd-debug.txt](https://github.com/raspberrypi/picoprobe/files/8455655/openocd-debug.txt)

I'm sure I must be doing something dumb.

dchokola, 2022-04-10:
I was right—I did something dumb. I did not see any documentation and assumed the SWD pins on the Picoprobe were used to program the other RP2040.

So others don't fall into the same trap, I've tabulated the pins below. Raspberry Pi team, please consider making a README.md containing the below table or similar.
| Pin | Function |
|---:|:---|
| 2 | SWCLK |
| 3 | SWDIO |
| 4 | UART_TX |
| 5 | UART_RX |
| 6 | RESET |

lurch, 2022-04-11:
> I did not see any documentation...

https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html contains a link to https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and Appendix A seems to have the wiring information you're looking for?

closed 2022-04-11 by aallan