raspberrypi/debugprobe #26: Opening serial port on picoprobe breaks target pico USB

marecl, 2022-05-14:
Hi
I started using picoprobe as my main programmer/debugger.
However I've encountered a bug, where if I open serial port on picoprobe, target shows as unknown device (cannot get device descriptor).
UART and SWD don't even need to be connected physically and happens only when COM port is open.
This does not occur when using external UART bridge.

Is this a fixable issue? I've been looking for a fix, but now the only solution is using an external converter.
Not a dealbreaker but contrary to Arduino, now I have to use 3 USB ports :(

JamesH65, 2022-05-14:
What OS are you using?

marecl, 2022-05-14:
Windows 10

Pobierz aplikację Outlook dla systemu Android<https://aka.ms/AAb9ysg>

mcuee, 2022-05-21:
I can not reproduce this issue under my Windows 11 laptop.
1) Build picoprobe firmware and upload it to the Pico
2) Pico is recognized by Windows. COM port assigned is COM21.
3) I use TeraTerm to open COM21
4) No ssues.

@marecl 
If you encountered such issues, you may want to try to add a HUB in between, especially external powered hub to see if that helps.

Can you try another PC to see if that helps?

marecl, 2022-05-21:
Okay, I have a clean Windows 10 install, so I will retrace my steps:
1. Install picoprobe interface 2 driver with Zadig - libusb 1.2.6.0
2. Install target pico driver
3. Connect them both
4. Open PuTTY for Picoprobe serial interface
USB breaks (Descriptor request failed)

> @marecl If you encountered such issues, you may want to try to add a HUB in between, especially external powered hub to see if that helps.
Connected target pico to different USB controller - it works.
Apparently pico is really picky when it comes to USB hubs. They don't work with my (powered) KVM, but work when connected to a hub in my monitor. Doesn't when both are on PCs front USB but do when one is connected to the back.
Weird but ok. Thanks @mcuee for a suggestion. Totally missed this one.

closed 2022-05-21 by marecl

lurch, 2022-05-21:
> Apparently pico is really picky when it comes to USB hubs. ... Doesn't when both are on PCs front USB but do when one is connected to the back.

See also the description of RP2040-E5 in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf , in case that's what you're running into.

Isysxp, 2025-08-14:
I know this is an old thread but, many thanks for the info as above as I have just had this problem as well
with the picoprobe and target plugged into the same USB hub. Found another USB port on my PC ... working!!!!