raspberrypi/debugprobe #28: Reserve an ADC pin for target voltage measurement

sourcebox, 2022-06-11:
Debug probes often read the target voltage via some ADC pin (e.g. ST-Link). It would be advisable to reserve such a pin for future use, e.g. GP26 for ADC0.

lurch, 2022-06-12:
I've never used an ST-Link, but my guess would be that it does that so that it can detect whether it's connected to a 5V or a 3.3V device? Which wouldn't really work with Pico, as its maximum ADC input voltage is 3.3V.
Also, picoprobe was only developed to be a debug probe for Pico / RP2040, we don't (officially) support using at as a debug probe for other devices.

sourcebox, 2022-06-12:
The ST-Link was just mentioned as an example of a debug probe with that functionality. What it effectively does is detecting if a target is connected at all and powered up with a suitable voltage. The ST-Link as an example is not 5V tolerant on ADC lines, so no switching between different voltages.

If no target voltage is detected, the probe should keep the SWCLK and SWDIO pins in high-impedance state to prevent any leakage current flow into an eventually unpowered target. For that reason, most debug probes also use serial resistors in the signal lines to limit potential current flow.

lurch, 2022-06-12:
Ahhh. If it's just a "is the target powered-on or unpowered" test, then I guess a digital IO pin could be used for that (assuming that both target and debug-probe are 3.3V devices), rather than an ADC pin?
I dunno much about low-level electronics, but would putting serial resistors in the signal lines reduce the maximum speed of the signalling?

ping @liamfraser for his thoughts (as he developed picoprobe originally).

sourcebox, 2022-06-12:
Yes, a digital pin would be sufficient for just on/off checking. But since ADC pins are unused anyway, you could use one for a more exact readout. With ST-Link e.g., the reading is printed on the OpenOCD console on startup. Not that this is essential, but can give some hint in case of malfunction.

The serial resistors reduce maximum speed, that's true. But they're typically around 100 ohms in this application and that will not have any serious impact at the speed the probe is operating on. There's also the possibility to use other protection mechanisms for real high speed, but that's up to hardware designers.

dpryan, 2022-08-01:
It would be also be good to use another pin to output SWDIO_DIR which is used to control the level-shifter direction to V<sub>target</sub>. This is what the Black Magic Probe does.
I imagine it is pretty easy to add that to the PIO state machine if someone knew what they were doing.

P33M, 2022-11-11:
As picoprobe now uses CMSIS-DAP as the interface driver, there's no command that will return this data to the host, or configure the target IO voltage. Any debug adapter variant that had level shifters and (divided-down) voltage sense would have to set this up in the call to DAP_Connect and do so automatically. Setting SWD output drive to be on a separate GPIO is basically trivial.

CZEMacLeod, 2023-02-20:
@sourcebox Serial resistors at 22 to 100ohms actually can prevent echo/ringing on long lines (like the debug cables) and can _increase_ the maximum speed in some scenarios. It is best to put them as close to the chip as possible for best effect.

lurch, 2023-02-22:
> It would be also be good to use another pin to output SWDIO_DIR

See #57 (if I've not misunderstood things!)

P33M, 2024-11-07:
This is out of scope for the hardware targeted in this repo. Boards with configurable level shifters need platform-specific code.

closed 2024-11-07 by P33M