raspberrypi/debugprobe #30: Picoprobe won't run on Adafruit Feather RP2040

jackw01, 2022-06-27:
I needed a second RP2040 board to use as a Picoprobe for SWD debugging another Raspberry Pi Pico so I went to Micro Center and purchased an Adafruit Feather RP2040 because they were out of Pi Picos. The board I got appears to be otherwise functional but will not connect to any computer over USB (tested on one Raspberry Pi 4 and two Windows PCs) when running the Picoprobe firmware. Nothing is displayed in lsusb on Linux or device manager on Windows when it is connected. I don't believe there is an issue with my build/install process because I can flash Picoprobe on my other Pi Pico and use it to debug code running on the Feather RP2040. My Pi Pico shows up correctly in lsusb and device manager on all of my computers when running Picoprobe. 

I posted this on the Adafruit support forum and the person who responded was unhelpful and claimed that the only issue is that some GPIO pins are unavailable (pins 2 and 3 are available, which is all Picoprobe needs) and did not address the issue that the board was not presenting itself as a USB device.

lurch, 2022-06-27:
The pre-compiled picoprobe firmware _might_ only work on a Raspberry Pi Pico board. For 3rd-party boards you'll need to compile your own picoprobe firmware from the sourcecode, specifying the appropriate `PICO_BOARD=`, as detailed in Appendix D of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

closed 2022-06-27 by jackw01

yahyatawil, 2024-01-26:
A small note: `PICOPROBE_USB_CONNECTED_LED` in `picoprobe/include/board_pico_config.h maybe` needs to be changed to `13` from `25` while building with `-DPICO_BOARD=adafruit_feather_rp2040` option.