raspberrypi/debugprobe #34: configure: error: no acceptable C compiler found in $PATH

anders-bergman, 2022-08-01:
I'm trying to build OpenOCD as described in the _[Getting Started With Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) > Appendix A: Using Picoprobe > Build OpenOCD > Windows_.

However, when running the command  `./configure --enable-picoprobe --disable-werror `, I get an error:
**configure: error: no acceptable C compiler found in $PATH**

Full output of the command:
```
$ ./configure --enable-picoprobe --disable-werror
configure: loading site script /etc/config.site
checking for makeinfo... makeinfo
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... no
checking for cc... no
checking for cl.exe... no
checking for clang... no
configure: error: in `/home/AndersBergman/openocd':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details
```

See [config.log](https://gist.github.com/anders-bergman/631baff7149120cb267a05332169a9a9) for the details.

My $PATH contains:
```
$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/opt/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
```

The documentation states that after running the pacman commands, one should:
> Close MSYS2 and reopen the 64-bit version to make sure the environment picks up GCC.

but this did not make any difference for me.

lurch, 2022-08-01:
The Windows instructions in that PDF have been tested as working by multiple people, so if something's not working then it sounds like you've perhaps missed out one of the steps? Can you try following the instructions again, making sure you carefully follow every step?

anders-bergman, 2022-08-02:
Thanks for the feedback @lurch . From a clean Windows installation, what are the steps you need to install prior to _Appendix A_? What I can recall, I did only install the steps from chapter _9.2 Building on MS Windows_ . Should that be sufficient?

lurch, 2022-08-03:
Note that Section 9.2 installs things for the "Developer Command Prompt" environment, whereas Appendix A installs things for the "MSYS2" environment; and my understanding (I'm not a regular Windows user) is that these environments are basically independent from each other. This is why it's important that you follow the instructions carefully. :slightly_smiling_face:

anders-bergman, 2022-08-05:
Thank you @lurch , it make sense that you should be able to go directly to Appendix A.

I have reproduced the issue in a new clean virtual machine following the instructions from the guide to the point.
1. Setup a new clean virtual machine with Windows 10 Pro 21H2.
```cmd
ver
Microsoft Windows [Version 10.0.19044.1826]
```
2. Download and install MSYS2 (msys2-x86_64-20220603.exe)
"Next, next" setup accepting all defaults.
When setup completes, tick "Run MSYS 64bit now.".

3. Following the guide, all commands are executed successfully up until the command `./configure --enable-picoprobe --disable-werror` as originally reported.

_Expected result_: No error messages output when running the configure command.
_Actual result_: Error messages at the end of the output:
```
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details
```

kjrandez, 2022-10-29:
I had the identical issue. I found that I had to run "**MSYS2 MINGW**" (not MSYS2) from the windows start menu. Typing gcc into the termainl is only found when running MSYS2 MINGW. Don't ask me why there should be two separate options. And no the instructions don't say to do that. It's possible something changed with how MSYS2 works since they wrote it.

P33M, 2023-02-15:
We will shortly ship precompiled builds for Windows, and largely deprecate instructions for building openocd locally.

closed 2023-02-15 by P33M