raspberrypi/debugprobe #41: pico w isn't supported?

pingufreak, 2022-10-15:
Hi,

I tried to compile picoprobe for the pico w. Anyways, the pins for LED and SWD have changed on the pico w. Is there already a common solution for this? Else I would change the picoprobe source to be compatible with the pico w.

`In file included from C:\pico\picoprobe\src\led.c:29:
C:\pico\picoprobe\src\picoprobe_config.h:68:2: error: #error PICO_DEFAULT_LED_PIN is not defined, run PICOPROBE_LED=<led_pin> cmake
   68 | #error PICO_DEFAULT_LED_PIN is not defined, run PICOPROBE_LED=<led_pin> cmake
      |  ^~~~~
C:\pico\picoprobe\src\led.c: In function 'led_init':
C:\pico\picoprobe\src\led.c:39:15: error: 'PICOPROBE_LED' undeclared (first use in this function); did you mean 'PICOPROBE_H_'?
   39 |     gpio_init(PICOPROBE_LED);
      |               ^~~~~~~~~~~~~
      |               PICOPROBE_H_
C:\pico\picoprobe\src\led.c:39:15: note: each undeclared identifier is reported only once for each function it appears in
C:\pico\picoprobe\src\led.c: In function 'led_task':
C:\pico\picoprobe\src\led.c:49:18: error: 'PICOPROBE_LED' undeclared (first use in this function); did you mean 'PICOPROBE_H_'?
   49 |         gpio_put(PICOPROBE_LED, !((led_count >> LED_COUNT_SHIFT) & 1));
      |                  ^~~~~~~~~~~~~
      |                  PICOPROBE_H_
C:\pico\picoprobe\src\led.c: In function 'led_signal_activity':
C:\pico\picoprobe\src\led.c:55:18: error: 'PICOPROBE_LED' undeclared (first use in this function); did you mean 'PICOPROBE_H_'?
   55 |         gpio_put(PICOPROBE_LED, 0);
      |                  ^~~~~~~~~~~~~
      |                  PICOPROBE_H_
NMAKE : fatal error U1077: "C:\PROGRA~2\ARMGNU~1\1142DA~1.3RE\bin\AR19DD~1.EXE": Rückgabe-Code "0x1"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.33.31629\bin\HostX86\x86\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.33.31629\bin\HostX86\x86\nmake.exe"": Rückgabe-Code "0x2"
Stop.
`

Thx

Kind regards,

Pingu

kilograham, 2022-10-15:
yeah there is no regular LED on the Pico W; it would seem reasonable to allow building without a LED.

It would be a separate task to make a PicoW build that actually uses the onboard LED (which is accessed via the WiFi chip)

pingufreak, 2022-10-25:
I'm using a normal pico right now, anyways, it's really slow.

pawsaw, 2023-02-21:
Hi, what is the status of this? Lost many hours coming to this conclusion: picoprobe doesn't work on pico w.

It would be great, either to support pico w or update the documentation.

P33M, 2023-08-15:
> @P33M Does the `picoprobe.uf2` at https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.0.3 work on the Pico W, or should there be a picow-specific UF2 file there too?

No, flashing that will result in nonfunctional wireless. It needs a board include file similar to debugprobe.

meltdown03, 2023-08-15:
> > @P33M Does the `picoprobe.uf2` at https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.0.3 work on the Pico W, or should there be a picow-specific UF2 file there too?
> 
> No, flashing that will result in nonfunctional wireless. It needs a board include file similar to debugprobe.

What ends up without wireless, the debug target board? I only have one Pico W so I cant check how it debugs another Pico W.

Ziothh, 2023-11-27:
So erm I compiled picoprobe with `cmake --DPICO_BOARD=pico_w .. && make` (needs to be one tick instead of two) and when I flash `/build/picoprobe.uf2` to my Pico W,  the LED still doesn't light up. 

Since it doesn't need the WIFI functionality the LED should be able to work. Am I missing something?

lurch, 2023-11-28:
On the Pico W, the LED is controlled by a GPIO on the WiFi chip, rather than by an LED on the RP2040 chip (which is why there's separate blink example-code for [Pico](https://github.com/raspberrypi/pico-examples/tree/master/blink) and [PicoW](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/blink)). And it sounds like PicoProbe doesn't have code to explicitly support the PicoW's LED?

However the good news is that PicoProbe **does** have explicit support for the [DebugProbe](https://www.raspberrypi.com/products/debug-probe/).

yahyatawil, 2024-01-26:
Just for confirmation. I've used `-DPICO_BOARD=pico_w` and uploaded the new .uf2 to Pico W. **I was able to run the Openocd.** 

Note: For some reason I can hear a resonance from the wifi module when my ear is close to the board.

fpaterra, 2024-02-29:
I'm having this same problem. I've tried different Pico and I'm able to start OpenOCD with picoprope loaded on a non-W pico connected to a non-W pico but when I attempt to start with picoprobe loaded on a non-W pico attached to a pico-w, OpenOCD fails with ```Error: Failed to connect multidrop rp2040.dap0```. If I run with the -d flag the specific failures I get are 

```
Error: 709 539 adi_v5_swd.c:328 swd_connect_multidrop(): Failed to connect multidrop rp2040.dap0
Debug: 710 539 command.c:529 exec_command(): Command 'dap init' failed with error code -4
User : 711 539 command.c:601 command_run_line():
Debug: 712 539 command.c:529 exec_command(): Command 'init' failed with error code -4
```

Thanks in advance for any help!

lurch, 2024-03-01:
@fpaterra AFAIK that should work fine - have you double-checked your wiring?

fpaterra, 2024-03-01:
> @fpaterra AFAIK that should work fine - have you double-checked your wiring?

Thanks for your reply @lurch. I have doubled checked and had other look over them as well. I also tried switching out the specific pico boards used both the picoprobe host as well as the pico w target. I'm sure I have some thing wrong but I've spent hours on this have not been able to gain any insight. Is there more debugging action I can do to help identify what is wrong?

Thank you again for your help!

fpaterra, 2024-03-01:
Interesting. When I connect a Pico Debugger Probe (https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) and run the same openocd command:
```$ openocd -d -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"```

I get a similar error output, differing only in line numbers. Does this help? Any ideas?

```
Error: 685 532 adi_v5_swd.c:328 swd_connect_multidrop(): Failed to connect multidrop rp2040.dap0
Debug: 686 532 command.c:529 exec_command(): Command 'dap init' failed with error code -4
User : 687 532 command.c:601 command_run_line():
Debug: 688 532 command.c:529 exec_command(): Command 'init' failed with error code -4
User : 689 532 command.c:601 command_run_line():
```

I should also note that I get a series of repeating debug messages before these errors (like 10s or more of them) saying this:

```
Debug: 682 529 cmsis_dap.c:1257 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 683 530 cmsis_dap.c:1272 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 684 531 cmsis_dap.c:946 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
```

I'm grateful for any help!

fpaterra, 2024-03-01:
OK more in the saga. On a whim I attached the USB to power and now things are working using the pico debugger probe. I had tried that with the pico probe software on a non-w pico as well but it did not help. Hopefully I'm up and functional now.

HyperNiki, 2024-03-13:
@fpaterra , Hello, I'm encountering the same issue with the Pico Probe. Have you had any success in resolving it? I attempted to address it by installing various versions of openoсd that align with the picoprobe versions (mostly around the same date), but unfortunately, the error persisted. The error message consistently appeared as 'DAP init failed' or 'Failed to connect multidrop rp2040.dap0,' along with the debugging information 'SWD ack not OK @ 0 JUNK' that you previously mentioned. I also made an effort to launch an older version of Ubuntu 18.04, installing both the old versions as well as new versions of openoсd and picoprobe in separate attempts, but I encountered the same error each time.

What's more, there were no such errors two months ago, and the debugger was launching successfully. It's puzzling how everything has changed now.
I would be very grateful for any help

fpaterra, 2024-03-13:
I was not able to get it to work, I think something changed and there is an incompatibility with PicoProbe on a pico (non-w or w) targeting a pico-w. I moved to using the Pico Debug Probe hardware device and it worked immediately so I would recommend buying one of those (~$25).

lurch, 2024-03-13:
> On a whim I attached the USB to power and now things are working using the pico debugger probe.

For the benefit of other people who might run into the same problem, could you explain in more detail what you mean by "I attached the USB to power" please?

hinxx, 2024-07-22:
FWIW, it works for me; I can use pico as a probe and picow as a target and vice versa. Wiring is as per [Appendix A: Using debugprobe](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

I added a `include/board_picow_config.h`, which is a copy of `include/board_pico_config.h` with the `#define PROBE_USB_CONNECTED_LED 25` line commented out. Further more the check for `DEBUG_ON_PICOW` was added to `src/probe_config.h` where it deals with including the `include/board_picow_config.h`, and `CMakeLists.txt`.

Basically I searched for `DEBUG_ON_PICO` and added newly introduced `DEBUG_ON_PICOW`. The only difference between the two is not using LED on picow.

OTOH, I guess adding the support for the picow LED would be straightforward ; taking the code from picow blink example.

ignatious-p, 2025-04-26:
> FWIW, it works for me; I can use pico as a probe and picow as a target and vice versa. Wiring is as per [Appendix A: Using [...]

Thanks @hinxx! Your method works for me too.

However, I also found that uploading the same firmware to the Pico W also _Works Just Fine™_ (although it is **wrong** to do so), since the Pico and Pico W have the same pinout (except for the LED pin). 

> Interesting. When I connect a Pico Debugger Probe (https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) and run the same openocd command: `$ openocd -d -f [...]

@fpaterra I got the same issue. I resolved it by following the instructions on connecting the Pico being debugged to the Pico debugger as decribed in [Appendix A: Using debugprobe](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

Therefore, I can say that you are seeing these messages due to:
1. Not connecting up the boards correctly.
2. Not powering on the device being debugged.

Hope this helps you or anyone in the future!