raspberrypi/debugprobe #42: Speed

pingufreak, 2022-10-25:
Flashing an elf through gdb takes aproximately 2 minutes, which is too long compared to the Raspberry Pi 4 SWD or Blackmagic Probe flashing. Is there any improvement ongoing? Also to use PIOs for faster flashing? Thx

lurch, 2022-10-28:
You might want to have a look at #31 ?

flaviut, 2022-10-29:
Following up here as well @lurch, but #31 does not provide any performance improvement until the message queue is lengthened from 2 to 255. Additional details in the patch here: https://github.com/raspberrypi/picoprobe/pull/31#issuecomment-1295517910

This is because a short message queue runs against USB's minimum 2ms latency, which at 8 bytes/roundtrip is very slow.

P33M, 2023-02-21:
Please quantify "speed".

closed 2023-09-21 by P33M