raspberrypi/debugprobe #44: Add JTAG support

P33M, 2022-11-11:
Given that we already swap probe programs in and out of PIO on the fly, it would be easy enough to add JTAG support.

How does PIO assist in this scenario - is the DAP protocol sufficiently high-level that we could accelerate scan-IR and scan-DR ops?

diggit, 2022-12-21:
DAP JTAG API is pretty low level.
https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__JTAG__Sequence.html
PIO could handle this configurable 1~64 bit "SPI".

Patronics, 2024-03-03:
Worth noting that there seem to be a few other projects that implement JTAG on the Pi Pico, perhaps some of that functionality could be ported over relatively easily?

There's https://github.com/phdussud/pico-dirtyJtag/

https://github.com/kholia/xvc-pico/

and https://git.lain.faith/sys64738/DragonProbe.git

Himanshu-4, 2026-04-25:
i had developed the JTAG support in existing pi pico firmware, i have tested on esp32 and working perfectly. 
the pio version of JTAG is still under development.
You can check it out on https://www.linkedin.com/feed/update/urn:li:activity:7436369988216569856/