raspberrypi/debugprobe #45: pico Timers not counting when debugging with picoprobe-cmsis-v1.0.1 and upstream openocd 0.12.0-rc2

BobyClaws, 2022-11-16:
## My Raspberry Pi Pico's timers are not working the counter is stuck at zero after switicing to the latest release of pico

### When It used to work:

followed usual guide from pico c/c++ sdk to compile openocd and picoprobe
have used openocd 0.11.x and the latest uf2 provided (one without cmsis-dap support)

my vscode launch.json 

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Pico",
            "request": "launch",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/target/thumbv6m-none-eabi/debug/learn_pico",
            "preLaunchTask": "build_binary",
            "servertype": "external",
            "gdbPath" : "gdb-multiarch",
            "gdbTarget": "localhost:3333",
            "runToEntryPoint": "main",
            "preLaunchCommands": [
                "monitor init",
                "monitor reset init",
                "monitor halt",
            ],
            "postRestartCommands": [
                "break main",
                "continue"
            ],
        }
    ]
}
```
timers are working correctly

### When it stopped working

with upstream openocd 0.12.0-rc2 (build steps followed: $ ./bootstrap $ ./configure $ make -j4)
and picoprobe-cmsis-v1.0.1

and with the following command to start openocd

`openocd -d2 -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c 'adapter speed 10000;'`

My timers' counter no longer counts up.

`monitor reset` command makes them tick again. but 

upon using any thing else like in above vscode config preLaunchCommands, they stop counting.

P33M, 2022-11-16:
Two things have changed in your setup. Which was the breaking change? `picoprobe-cmsis-v1.0.1` will work on both downstream openocd and upstream.

P33M, 2022-11-16:
When are you expecting the timers to increment - do they run when the target isn't halted?

BobyClaws, 2022-11-16:
I expect the timer specifically the 64 bit counter to start counting as soon as the program starts. And the counter to stop when I  raise a halt. In both openocd the counter counts up  when you just do a reset via `monitor reset` and I can `monitor halt` and `continue` to stop and resume aswell.  but with upstream when I or vscode try to do anything like `monitor reset halt` or `monitor reset init` and then `continue`  my code just reports the counter value as 0 all the time.

here's a sample rust code: https://gist.github.com/BobyClaws/0e224d66b62b6a6a6d1bc680c1908395#file-main-rs-L73

cstoitner, 2022-11-26:
I have the same problem. With OpenOCD 0.12.0-rc2 and picoprobe-cmsis-v1.0.1 the timer hardware always returns 0 when gdb is connected. When running the same file without gdb, even if flashed with openocd over the picoprobe, the timer will start returning non zero values very quickly after starting main.

Example code:
```
#include <cstdio>
#include <pico/stdlib.h>
#include <hardware/structs/systick.h>

int main()
{
    stdio_init_all();
    systick_hw->csr = 0x5;
    systick_hw->rvr = 0x00FFFFFF;

    printf( "start...\n" );
    while ( true )
    {
        printf( "time: %llu systick: %lu\n", time_us_64(), systick_hw->cvr);
    }
}
```

The systick hardware works as expected.

cstoitner, 2022-11-30:
I got another CMSIS based probe to test. Turns out this is not a bug with the pico probe or openocd.

The default configuration for openocd 0.12rc2 with an rp2040 target will debug both cores with dedicated gdb instances. With my configuration this meant the second core was always stopped as I never connected the second gdb instance. The rp2040 timer is set up by default to stop counting when any core is stopped by a debugger and so the timer would never advance.

You can use `set USE_CORE 0` in the openocd config to select the first core only and the problem disappears. Multicore debugging with a single gdb instance is not yet supported with upstream openocd on the rp2040.

closed 2023-02-15 by P33M

dojoe, 2024-04-13:
> You can use `set USE_CORE 0` in the openocd config to select just the first core for debugging and the problem disappears. Multicore debugging with a single gdb instance is not yet supported with upstream openocd on the rp2040.

For others finding this issue, here are two details I tripped over at first that weren't immediately clear to me:
1. If you're launching OpenOCD manually, `set USE_CORE 0` must be set _before_ rp2040.cfg is loaded since the variable influences the behavior of rp2040.cfg. If OpenOCD only brings up a single gdb server according to its output, the setting worked correctly.
2. If the Pico still hangs after you changed the setting, core 1 might still be in debug mode from when OpenOCD was configured to control both cores. Power cycle or reset the Pico, that should clear things up.