raspberrypi/debugprobe #48: openocd: ailed to claim picoprobe interface

plebioda, 2022-11-25:
I am not able to use openocd with picoprobe on xubuntu:
```
~/pico/openocd (rp2040)$ ./src/openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g228ede4 (2022-11-18-18:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to claim picoprobe interface
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

The device is detected:
```
[  848.934684] usb 2-2: USB disconnect, device number 8
[  921.175453] usb 2-2: new full-speed USB device number 9 using ohci-pci
[  921.694255] usb 2-2: New USB device found, idVendor=2e8a, idProduct=0004, bcdDevice= 1.01
[  921.694261] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  921.694263] usb 2-2: Product: Picoprobe CMSIS-DAP
[  921.694265] usb 2-2: Manufacturer: Raspberry Pi
[  921.694266] usb 2-2: SerialNumber: E66160F4235DC136
[  921.710251] cdc_acm 2-2:1.1: ttyACM0: USB ACM device
```

And I have the 60-openocd.rules in /etc/udev/rules.d from openocd/contrib:
```
...
# Raspberry Pi Picoprobe
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", MODE="660", GROUP="plugdev", TAG+="uaccess"
...

Do I miss something? I followed the description refered in the readme file.

KagurazakaSetsuna, 2022-11-26:
Using interface/cmsis-dap.cfg will allow openocd to detect the picoprobe, it seems like the usb PID was changed in recent releases.

lurch, 2022-11-26:
Yes, we'll be releasing an updated version of the getting-started PDF soon.

bezim, 2022-12-05:
I had a similar issue, documenting my steps below in case it can help someone else. 

It would be great if the Pico team could modify the pico_setup.sh script, or whatever else is necessary, so that this works out of the box on a Raspberry Pi. THX!

**Setup**
Raspberry Pi 400 with _fresh_ image of Raspberry Pi OS flashed using the Raspberry Pi Imager.
Two Raspberry Pi Picos attached via USB, one is the target, one used for Picoprobe. Wiring connection according to Getting Started Guide pdf (GSG, build date 2022-11-30) figure 37 in appendix A.

**Steps taken**
Installed the SDK using the script from Chapter 1 of the GSG.
`pi@raspberrypi:~ $ ./pico_setup.sh`

Flashed picoprobe.uf2 to the debugger Pico.

Tried running openocd according to page 67 in appendix A of the GSG, but got the following error:
```
pi@raspberrypi:~ $ openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-12-04-20:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to find a matching CMSIS-DAP device
```

Copying 60-openocd.rules solved this:
`sudo cp /home/pi/pico/openocd/contrib/60-openocd.rules /etc/udev/rules.d/`

But now I got a clock error:
```
pi@raspberrypi:~ $ openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-12-04-20:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6609103C3543829
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
```

This was fixed by passing the adapter speed parameter:
```
pi@raspberrypi:~ $ openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -s tcl
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-12-04-20:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6609103C3543829
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

This finally allowed me to successfully upload the blink example:
`openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -c "program blink/blink.elf verify reset exit"`

lurch, 2022-12-05:
https://github.com/raspberrypi/picoprobe/issues/49#issuecomment-1331748165 says that further updates will be coming soon.

pranjalchanda08, 2023-07-06:
I tried all above but now facing: 
```sh
root@kurama:~# openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-07-06-12:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E661A4D4175B4F2D
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : DAP init failed

```
Says: `DAP init failed`

closed 2023-09-21 by P33M