raspberrypi/debugprobe #49: Using the new  cmsis-dap picoprobe is extremely slow.

mwinters-stuff, 2022-11-29:
I have tried the new release of pico-probe and was quite dissapointed..

Using VSCode, with Cmake and Cortex-Debug extensions.
From hitting the debug (F5) to the automatic breakpoint at the main function:

The "elf" file is 497264 bytes.

using "picoprobe.cfg"  13 seconds. (picoprobe commit #33ed0e7)
using "cmsis-dap.cfg" 90 seconds. (picoprobe HEAD)

Step through the debug is also much slower.

I guess there is a bus speed configuration missing in the defaults somewhere, but at the moment that time difference makes the cmsis-dap un-usable.

P33M, 2022-11-29:
Kia Ora. Can you provide all the output from the vscode terminal tab that displays the openOCD log messages when starting a debug session? No need to include what happens during your debug session.

mwinters-stuff, 2022-11-30:
Hi.

I have been using my vscode devcontainer, https://github.com/mwinters-stuff/vscode-devcontainer-raspberrypi-pico

These are the logs from the project I am currently working on.

### OpenOCD Debug, using picoprobe
```
Cortex-Debug: VSCode debugger extension version 1.6.7 git(b0f5563). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from objdump-multiarch --syms -C -h -w /workspaces/pico-joystick/build/dev_hid_composite
Reading symbols from nm-multiarch --defined-only -S -l -C -p /workspaces/pico-joystick/build/dev_hid_composite
Launching GDB: gdb-multiarch -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Error: nm-multiarch failed! statics/global/functions may not be properly classified: Error: spawn nm-multiarch ENOENT
    Expecting nm next to objdump. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn objdump-multiarch ENOENT    ENOENT means program not found. If that is not the issue, please report this problem.Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /workspaces/pico-joystick -f /home/vscode/.vscode-server/extensions/marus25.cortex-debug-1.6.7/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 16 ms
Finished reading symbols from nm: Time: 13 ms
da_3 () at /apps/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S:253
253	 bmi da_6
Program stopped, probably due to a reset and/or halt issued by debugger
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
Note: automatically using hardware breakpoints for read-only addresses.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00

Thread 1 hit Temporary breakpoint 4, main () at /workspaces/pico-joystick/src/main.cpp:46
46	int main(void) {

```

### Log from gdb-debug/openocd terminal tab for picoprobe
```
[2022-11-30T00:37:53.912Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /workspaces/pico-joystick -f /home/vscode/.vscode-server/extensions/marus25.cortex-debug-1.6.7/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-11-30-00:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
[... 198 lines cut ...]
        "interface/picoprobe.cfg",
        "target/rp2040.cfg"
        ],
      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
      "runToEntryPoint": "main",
    },
    {
      "name": "Cortex Debug CMSIS-DAP",
      "cwd": "${workspaceRoot}",
      "executable": "${command:cmake.launchTargetPath}",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "gdbPath": "gdb-multiarch",
      "device": "RP2040",
      "configFiles": [
        "interface/cmsis-dap.cfg",
        "target/rp2040.cfg"
        ],
      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
      "runToEntryPoint": "main",
    }    
  ]
}
```

P33M, 2022-11-30:
```
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
```

This is why - no interface speed has been specified. Add `adapter speed 5000` to openOCDLaunchCommands which should work. The getting-started guide and our example launch.json file will be updated in due course.

mwinters-stuff, 2022-12-04:
Aaah, Thanks - I am not familiar with CMSIS-DAP, being the first time I have used it. The time to main was 9.8 seconds, and 8.6 if I upped the speed to 10,000 

Closing the issue.
:)

closed 2022-12-04 by mwinters-stuff