raspberrypi/debugprobe #53: Discussion: SWCLK speed

rgrr, 2022-12-05:
It has already been mentioned, that clock speed SWCLK does not correspond to the setting done via openocd ("adapter speed"): https://github.com/raspberrypi/picoprobe/pull/47

The wrong calculation happens in sw_db_pio.c:
```
#define MAKE_KHZ(x) (CPU_CLOCK / (2000 * ((x) + 1)))
```

Parameter "x" is DAP_Data.clock_delay and calculated in DAP.c which takes DELAY_SLOW_CYCLES and IO_PORT_WRITE_CYCLES into account.

I'm wondering if those parameters are actually of interest and for the picoprobe the setup of the SWCLK is more relevant.

In this case the calculation in MAKE_KHZ() could revert the DAP.c one.  Definition would then look like
```
#define MAKE_KHZ(x) ((CPU_CLOCK / 2000) / (delay * DELAY_SLOW_CYCLES + IO_PORT_WRITE_CYCLES))
```

Any opinions about this?  If not, I would create a PR for this.

P33M, 2022-12-05:
MAKE_KHZ is fakery designed to get around the fact that CMSIS-DAP tries to be clever and passes you a value to be used inside a for() loop, not an actual clock frequency. We just lie about DELAY_SLOW_CYCLES and IO_PORT_WRITE_CYCLES, because PIO has an actual clock divider that directly sets SWCLK. If the lies are wrong, then the amount of fakery should be adjusted so that the number at the end of the calculation is correct.

rgrr, 2022-12-28:
I understand, that this is just some doing as if... but I think my cheating is better because it does not surprisingly increase the PIO frequency by a factor of two or so.  Additionally the corresponding PR #54 checks a maximum frequency so the PIO frequency should be never out of range.

P33M, 2023-03-02:
I can't replicate this with the most recent release and openocd 0.12. Logic analyser confirms this, at least for frequencies up to 5000kHz.

rgrr, 2023-03-02:
Perhaps because for your settings of DELAY_SLOW_CYCLES and IO_PORT_WRITE_CYCLES?

My code snippet above is nonsense BTW, the parameter must be (delay) of course.

If you're happy with your current solution you could close this issue.

closed 2023-09-21 by P33M