raspberrypi/debugprobe #55: Picoprobe-cmsis-dap doesn't appear in `/dev` on macos

TheButlah, 2022-12-15:
On version 1.0, if I check `/dev` before and then after my probe is connected:
```
❯ ls /dev > a
❯ ls /dev > b
❯ diff a b
16a17
> cu.usbmodem2102
214a216
> tty.usbmodem2102
```

However, on version 1.0.1, I see no output from the diff. Additionally, the CMSIS-DAP debugger I'm using (probe-rs) detects the v1.0 firmware but not the v1.0.1 firmware.

In both cases, I am testing this with *just* the rp2040, with no "chip being tested" connected.

P33M, 2022-12-15:
On what hardware, operating system?

Do you have any evidence that the 1.0.1 firmware was loaded successfully? Are there any system log messages?

TheButlah, 2022-12-15:
This was on macos 12.6, apple m1 processor. There was no chip other than the pico probe itself connected. USB cable was validated to work already, as I could flash the uf2 via it.

I am not sure what to run to check system logs, which is why I did the `ls /dev` approach. Let me know which commands to use and I'll run them! I see nothing in `/var/log/system.log` fwiw, but I'm not sure that is helpful because even when the UF2 drive appears, nothing gets updated in that logfile. So there must be another way to debug usb on mac.

I just did the experiment again. Flashing the 1.0.1 firmware seems to flash successfully (I assume), but after a reset the board does not appear to my CMSIS-DAP compliant debugger.

Flashing the v1.0 firmware, it *does* appear.

What

liamfraser, 2022-12-19:
Can you try with openocd? I am on MacOS 13.1 with head of OpenOCD and head of Picoprobe: It works fine.

```Open On-Chip Debugger 0.12.0-rc2+dev-00998-g77c281d2d (2022-12-19-15:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6616407E384AB2A
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz```

TheButlah, 2022-12-21:
I believe this was a red herring. IT works now. The KB2040 board I'm using requires me to reset a bunch of times to get it to appear to serial, which may be a hardware design issue, or maybe my mac drivers are bad. Either way, the same behavior happens on both versions of the picoprobe firmware, so I'm closing this issue.

closed 2022-12-21 by TheButlah