raspberrypi/debugprobe #56: Can't find a picoprobe device on Ubuntu 22.10.

overheat, 2022-12-15:
I follow "Appendix A: Using Picoprobe" in the Pico Getting Started Guide to use a Raspberry Pi Pico debug another Raspberry Pi Pico W (with Wi-Fi).

But got this error:
```
$ openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-12-15-19:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```
```
$ lsusb 
...
Bus 001 Device 002: ID 2e8a:000c Raspberry Pi Picoprobe CMSIS-DAP
...
```

```
$ ls -l /dev/bus/usb/001/002
crw-rw----+ 1 root plugdev 189, 1 12月 15 19:36 /dev/bus/usb/001/002
```

P33M, 2022-12-15:
You are using an old version of the PDF.  Invoke openocd with `openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl`. You may or may not need Sudo access depending on your Udev install.

lurch, 2022-12-15:
Perhaps you're looking at an old version? It [now says](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)
```console
$ src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl
```

overheat, 2022-12-15:
> ```
> src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl
> ```

Thank you all for quickly reply.

`$ src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl` did change output:
```
$ src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-12-15-19:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E660D4A0A7899A35
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
```

> You are using an old version of the PDF. Invoke openocd with `openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl`. You may or may not need Sudo access depending on your Udev install.

my Udev rules related to picoprobe:
```
# Raspberry Pi Picoprobe
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", MODE="660", GROUP="plugdev", TAG+="uaccess"
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="000c", MODE="660", GROUP="plugdev", TAG+="uaccess"
```
in `/etc/udev/rules.d/60-openocd.rules`

JamesH65, 2022-12-15:
Are you plugged in to a debugging target? And did you try with sudo?

overheat, 2022-12-15:
Yes, wires! 
I just connect two pico's SWD together. What stupid!

I notice that PDF just update, but still missing `adapter speed 5000`. You can add it in the end of `$ src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl`.

Thank you all, guys!

closed 2022-12-15 by overheat

lurch, 2022-12-15:
> but still missing `adapter speed 5000`

Rest assured that that will be included in the next release of the Pico databook PDFs; but we're not sure yet when that'll be.

topherbuckley, 2023-01-16:
Can someone explain why the `-c "adapter speed 5000"` became necessary? Can this not be set in some configuration file to save the typing every time? It also make the VS Code setup more cryptic trying to find a way to append this to automated code.

overheat, 2023-01-23:
Can someone explain why the -c "adapter speed 5000" became necessary?
Need this because configuration file miss this.
Can this not be set in some configuration file to save the typing every time?
Can be

d-adamson, 2023-01-27:
Using Linux Mint 21.1
The following worked for me 
Made a backup of the original file (just in case)
$ cd  ~/pico/openocd/tcl/target/
$ cp p2040.cfg p2040.cfg.original

Edited rp2040.cfg and added the following at the bottom of the file & save
adapter speed 5000

The command now works
$ openocd -s ~/pico/openocd/tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg

P33M, 2023-01-27:
> Can someone explain why the -c "adapter speed 5000" became necessary? Need this because configuration file miss this. Can this not be set in some configuration file to save the typing every time?

Interface and target files don't and cannot take into account wiring between the debug target and probe. Long or poor connections will significantly degrade the maximum achievable speed, so the selection of swclk frequency should be explicit.

topherbuckley, 2023-02-03:
Shall we make a new issue for this? As this topic is slightly tangential to the original topic and I feel this still is not "resolved". 

I think @d-adamson 's approach is not ideal as you're editing the cmsis-dap.cfg rather than the pico.cfg file that used to work. It would seem to me that the cmsis-dap.cfg file is intended as a more generalized interface and the pico.cfg file needs to be updated in similar ways as has been done to the cmsis-dap.cfg and then this line added to that file. 

@P33M , yes, but I would argue there should be some reasonable default value (with the ability to be explicitly changed when necessary via a flag/input parameter as is currently done), otherwise you could make the same argument for all other parameters as well.

P33M, 2023-02-03:
There is an outstanding change to the getting-started guide that has not yet been published, which will include this parameter.

topherbuckley, 2023-02-08:
@P33M is there no plan to revise the pico.cfg then? Can't say I understand the .cfg files enough to do anything about it myself yet, but reverting a default value and requiring an additional parameter seems like a temporary workaround rather than a long term solution. I understood your logic about wire length, but then why not just have a default at a reasonably low speed and those with pcb traces or whatever optimal signal path can increase it with the cl parameter? Wasn't this how it was when pico.cfg was used in the past?

P33M, 2023-02-08:
The config file will not be changed as it's an unmodified upstream file. If the instructions were correct, there wouldn't be an issue.

topherbuckley, 2023-02-10:
Well I think I have a simple fix for a PR I could make over on the openocd fork repo then. I'll ask this over there as well, but I see a bunch of related sounding branches (e.g. rp2040-master, picoprobe_tmp, picoprobe, rp2040-v0.12.0, etc.). Where should current PRs be based upon. @P33M it looks like you merged something on there 3 weeks back on to the rp2040 branch. Should I base the PR there?