raspberrypi/debugprobe #64: Unable to checkout 'a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4' in submodule path 'CMSIS_5'

topherbuckley, 2023-01-12:
Looks like there is some issue checking out the current commit for CMSIS_5 when executing the following line from the "Build and flash picoprobe" section of the User Guide:

When I run `git submodule update --init` it ends up on branch `master` with a dirty working directory. If I enter the `CMSIS_5` directory, and `git reset --hard` and remove any untracked files `git clean -xfd`, I end up on the `develop` branch. If I try to checkout a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4 with `git checkout a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4` I get the following error:

```
git-lfs filter-process: 1: git-lfs: not found
fatal: the remote end hung up unexpectedly
```

After installing git-lfs as per the [README](https://github.com/git-lfs/git-lfs#downloading) and repeating the 

```
git reset --hard
git clean -xfd
git checkout a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4
```
I am in a detached head at a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4.  Proceeding to build picoprobe from this point seems to work as expected. 

I'm guessing this is outside the control of this repo but I didn't see this documented anywhere, so hopefully this helps others who don't have git-lfs by default like I did.

neonmik, 2023-03-03:
Same issue here.

lurch, 2023-03-04:
I've just tried this myself, following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and it seemed to work fine? And I didn't need to install git-lfs, so I dunno quite what's going on there.

After cloning this repo, and then doing `git submodule update --init`, `git status` says:
```
On branch master
Your branch is up-to-date with 'origin/master'.

nothing to commit, working tree clean
```

and `git submodule status` says:
```
 a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4 CMSIS_5 (5.0.0-Beta4-2214-ga65b7c9a3)
 2dfdfc4ba4d8bb487c8ea6b5428d7d742ce162b8 freertos (V10.4.0-kernel-only-230-g2dfdfc4ba)
```

and `git lfs` (just to verify that I don't have it installed) says:
```
git: 'lfs' is not a git command. See 'git --help'.

The most similar command is
	log
```

Are you still having problems? What OS are you using?

topherbuckley, 2023-03-06:
I'm on Ubuntu 22.04, but there have been a few commits to the CMSIS_5 submodule since opening this, so maybe something changed upstream to resolve this? 

What about git version? I'm on 2.34.1 if it matters. Maybe lfs is handled differently between git versions? If @neonmik is still having issues, we can keep this open, otherwise feel free to close, as I provided a workaround in my first comment.

synthsandstuff, 2024-05-08:
After following the instructions in the getting started with pi pico document I ended up with the same issue on macos. Installing git-lfs seems to have solved the issue for me, after recloning the repo.

closed 2024-11-07 by P33M