raspberrypi/debugprobe #67: VS Code Debug - openocd listening on wrong port

d-adamson, 2023-01-27:
Dev System Linux Mint 21.1
Debugging via picoprobe

I've managed getting gdb and openocd to work together from the command line, and can set break pts, step through the code, etc. :+1: 
Had to modify
~/pico/openocd/tcl/target/rp2040.cfg
by adding a line at the bottom
adapter speed 5000

The following command now works:
$openocd -s ~/pico/openocd/tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg

gdb-multiarch and openocd are successfully communicating over localhost:3333

***************************
**However**, setting up the debugger in VS Code is not going so well. :-1: 
When VS Code launches the debugger, gdb-multiarch times out waiting for a connection to openocd. Inspecting the terminal window shows that vs code sends the following command to launch openocd

$/usr/local/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f ~/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg

This command specifies gdb_port 50000. However, I cannot find where this is being set. There is nothing obvious in settings.json & launch.json.  
By the way, if I paste the above command into a terminal and modify the line to gdb_port 3333 then openocd and gdb_multiarch are happy to communicate again.

Attached are the vs code json files.

A little help, please...

[vscode_json.zip](https://github.com/raspberrypi/picoprobe/files/10523736/vscode_json.zip)

d-adamson, 2023-01-28:
Adding onto my previous post. Seem to have made some progress, but still not fully working.
openocd commands can be passed in the launch.json file with the following syntax:

"openOCDLaunchCommands": ["adapter speed 5000", "gdb_port 3333", "init", "reset init"],

This allowed openocd and gdb to communicate briefly, then gdb-multiarch shuts down.

P33M, 2023-01-28:
This repository is for the firmware loaded on the debug probe, not for interactions between gdb and vscode. This question is best asked on the forums - https://forums.raspberrypi.com/viewforum.php?f=144

closed 2023-01-28 by P33M