raspberrypi/debugprobe #68: PicoProbe wont work, comes up as stuff other than interface 0 and 2 using the precompiled uf2 f

FJay81, 2023-02-20:
[image]
[image]
[image]
[image]

P33M, 2023-02-21:
Whose instructions are you following?

The releases in this repo do not require Zadig.

FJay81, 2023-02-21:
[Tutorial](https://www.digikey.co.uk/en/maker/projects/raspberry-pi-pico-and-rp2040-cc-part-2-debugging-with-vs-code/470abc7efb07432b82c95f6f67f184c0)
This one by digi key,
Can you use the pico for just uart without swd?

aallan, 2023-02-21:
See Appendix A of "[Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)" for the official tutorial.

FJay81, 2023-02-21:
[image]

So that will allow me to use the Pico like this thing?

aallan, 2023-02-21:
> So that will allow me to use the Pico like this thing?

So that needs fixing! 🤷‍♂️

FJay81, 2023-02-21:
i found something that does that now [The thingy](https://github.com/Noltari/pico-uart-bridge)

lurch, 2023-02-21:
See also https://www.raspberrypi.com/news/raspberry-pi-debug-probe-a-plug-and-play-debug-kit-for-12

lurch, 2023-02-21:
> So that will allow me to use the Pico like this thing?

Yes, you can use a Pico running PicoProbe as a USB-UART adaptor. The bit that "needs fixing" that @aallan is referring to is that the GND wire in that diagram is connected to the wrong hole on the breadboard.

FJay81, 2023-02-21:
ive seen the probe thingy, it was out of stock when i went to buy it hence this

aallan, 2023-02-21:
> ive seen the probe thingy, it was out of stock when i went to buy it hence this

Launched yesterday. Initial stocks at resellers were exhausted, but it'll be back in stock very soon. Plenty in channel.

FJay81, 2023-02-22:
So I got it sorted but when I went to connect to a router of mine which I checked with a multimeter to figure which pins are what it didn’t work, I’m using putty and a compile version of Pico probe from here and cannot for the life of me get an output in putty, I’m connected to the serial port whic comes up as usb cdc com7 and connect to it as com7 and different bauderates, but no output on the console is there any way to get an output?

lurch, 2023-02-22:
> when I went to connect to a router of mine

You won't be surprised to hear that that's not a use-case we can officially provide support for :wink:  You could try asking for help on the https://forums.raspberrypi.com/ ?

FJay81, 2023-02-22:
Ok will do

closed 2023-02-23 by aallan