raspberrypi/debugprobe #71: UART stops receiving after timeout

bofh69, 2023-02-24:
Running with the firmware that was shipped with the Pi Debug Probe (USB bcdDevice 1.01), the UART to USB seems to timeout and stops receiving after about two seconds.

My test setup is an x86 computer running Ubuntu 22.04, the PDP UART connected to an nRF52833 development kit. The DK and the PDP are connected to the same USB hub and that is connected to the computer.

I can then see serial output from the DK on the DK's debug UART, but not on the PDP's UART.

I then send something to the board through the PDP, the red led next to the USB connector lights up together with the green and yellow leds and after that I start to get the serial output from the DK. As long as serial data is sent, the green led keeps flashing and the red led remains on.

If the serial output is quiet for about two seconds, the red led goes off and after that further serial output is no longer sent. The green led also stops blinking.

This issue prevents it from being used as an USB UART.

P33M, 2023-02-24:
- Is the debug probe still listed in `lsusb` after the red LED goes out?
- What are the contents of the file `/sys/module/usbcore/parameters/autosuspend`?

bofh69, 2023-02-24:
@P33M 

The tty connection remains open and the device remains connected.

autosuspend was 2. Setting it to -1 to disable autosuspend makes it work correctly. Should that be necessary? It isn't with any other USB UART I've used.

P33M, 2023-02-24:
With an open file descriptor, the interface should still be considered "active" by usbcore, so that is unusual. I've not had Raspberry Pi OS autosuspend the probe.

There probably is a bug around suspend/resume (as I've not explicitly tested it).

P33M, 2023-05-02:
I think I've managed to reproduce what your machine is doing - if I enable autosuspend for the device by setting `/sys/bus/usb/devices/<id>/control` to `auto` then the LED turns off after two seconds. If I run `lsusb -v` this causes a wakeup.

My guess is that I need to signal resume when I have characters in the UART...

P33M, 2023-05-03:
I've spent some time playing with autosuspend on a Pi and I must recommend that you disable this behaviour in your distro (something is forcibly enabling it, our Debian-like defaults on Raspberry Pi OS do not do this). Apart from resume signalling needing to be implemented in software, there's a large amount of lag time (up to 300ms) between the bus returning to a functional state and the first IN token for the data endpoint. This is due to 
a) the hub status endpoint only being polled every 256ms
b) Linux recognising the resume event, and waking the hub workqueue (and any other internal processing)
c) resubmitting URBs for previously active endpoints
All of these delays are variable and additive. Many hundreds of characters could be transmitted in this time, and there is only so much buffering in the CDC implementation. Therefore the chance of data loss is high.

P33M, 2023-05-03:
In the absence of fully-tested suspend/resume support, a possible workaround is to make Linux more reluctant to put the device in suspend in the first place. I've pushed a branch here that doesn't advertise REMOTE_WAKEUP. In my tests, the device stays active after using it.

https://github.com/raspberrypi/picoprobe/tree/nowake

crazybolillo, 2023-08-03:
Having the same issue. I am running it on a normal Pico (not the debug probe), the green LED goes off and I won't longer be able to read the UART output through the CDC ACM interface.

crazybolillo, 2023-08-03:
Just started using the nowake branch right now and so far the LED stays green and I no longer have issues reading UART.

remcycles, 2023-08-22:
I had the same issue and upgrading to v1.0.3 (with 6473166494ebef2db9adc806e49471c63e38510b merged in) resolved it for me on Manjaro Linux.  Thanks!