raspberrypi/debugprobe #75: Can Picoprobe debug STM32 in MDK？

Hemistic, 2023-03-12:
Hello, I have a pico development board and I want to debug STM32 in Keil5, is that possible? I tried a lot, but it doesn't like a typical DAPLink. What setting should I change in the firmware?

P33M, 2023-03-12:
Have you tried following Keil's instructions? https://developer.arm.com/documentation/101451/0100/Configure-Debugger-Driver

kewxke, 2023-03-23:
keil mdk5 will not recognize the pi debug probe by default, even if I have assigned the default cmsis-dap.inf in keil mdk directory. So it means keil mdk5 pack cmsis dap driver is not compiled for Pi debug probe.

P33M, 2024-03-18:
Works for me in MDK 5 with an NXP and STM32 MCU, following Keil's instructions

closed 2024-03-18 by P33M