raspberrypi/debugprobe #76: What's happening when Pico Probe is SWD connected to a Pico, circuit is behaving differently...

kolbma, 2023-03-13:
I've connected an analog circuit which depends on differences in small microampere area to some Pins of the Pico.  
Without any connection on the SWD pins it works.  
As soon as I connect the Pico Debug Probe with the running Pico, there is some heavy difference happening on the circuit.
So I can not use it at the moment to debug in the productive environment. 
Something similar I've seen in my development environment when attaching/removing stuff to/from VBUS and mixing different potentials with same ground, that the debug connection crashed repeatably with reporting a not specified ARM error, but the Debugged Pico itself has no problem and continues. I can also restart the Debug session and it works until there is the next change on bus, like touching ground with with a voltmeter.

bjonnh, 2023-04-02:
Do they both share the same ground? How is your Pico powered? If it is not powered by the same computer that drives your picoprobe you may simply have ground loops.

kolbma, 2023-04-03:
And the Debug Probe isn't designed to handle this?

P33M, 2023-04-11:
>And the Debug Probe isn't designed to handle this?

It's not a fault of the Debug Probe. When RP2040 is driving SWDIO, which is usually connected to a long wire, VDDIO will bounce which will couple into other parts of the circuit. SWD is actively polled while a debug session is active.

If you're measuring microampere differences, you need to employ very strong noise filtering in the analogue sections of your circuit.

closed 2023-04-11 by P33M