raspberrypi/debugprobe #81: picoprobe - Interface 0 or 1 - zadig

NickDolfin, 2023-04-10:
I have a brand new Raspberry Pi Debug Probe (https://www.raspberrypi.com/products/debug-probe/)

I want to use it using visual studio code (https://arduino-pico.readthedocs.io/en/latest/platformio.html)

I find every where that I should use zadig to change the driver: "Remember that on Windows, you have to use [Zadig](https://zadig.akeo.ie/) to also load “WinUSB” drivers for the “Picoprobe (Interface 2)” device so that OpenOCD can speak to it."

(in https://arduino-pico.readthedocs.io/en/latest/platformio.html)

Using picoprobe.uf2:
I cannot find the Interface 2. Only 0 and 1:

[image]
[image]

I also found two different uf2 files: picoprobe.uf2 and debugprobe.uf2 in the raspi github repositories. 
The usb ID is different than described here: https://wiki.freepascal.org/ARM_Embedded_Tutorial_-_Raspberry_Pi_Pico_Setting_up_for_Development

Using debufprobe.uf2:
[image]

I cannot get it to work.
Questions:
Which uf2 is do I have to use?
Is the USB ID correct?

Anyone any idea?

P33M, 2023-04-11:
Those third-party instructions are incorrect. You do not need to use Zadig at all with the latest Picoprobe UF2.

Please follow the getting-started guide - https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

aallan, 2023-04-11:
See also https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html.

closed 2023-04-11 by aallan

NickDolfin, 2023-04-11:
I use platformio  - not the old fashioned arduino environment:
https://arduino-pico.readthedocs.io/en/latest/platformio.html

aallan, 2023-04-11:
> I use platformio - not the old fashioned arduino environment: https://arduino-pico.readthedocs.io/en/latest/platformio.html

We officially support our own [C SDK](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html), and [MicroPython](https://www.raspberrypi.com/documentation/microcontrollers/micropython.html). Unfortunately, we don't directly support either the Arduino environment or PlatformIO.

NickDolfin, 2023-04-12:
Aha,

There are there two differerent uf2 files in your repository.
Whats the difference between picoprobe.uf2 and debugprobe.uf2?

I changed the drivers and flashed one of the two uf2 files into the Raspberry Pi Debug probe. I like to restore it to the original version.
Can you make a screenshot of the zadig program showing the correct windows 10 drivers? And which uf2 file is standard flashed in the debug probe?

w4mmp-ron, 2023-04-12:
Hello,
As mentioned,  zadig is NOT used any longer for either Pico Probe or Debug Probe.   If you have the correct firmware loaded below is what windows will be display in the device manager.
[image]

NickDolfin, 2023-04-12:
Thanks, but that picture does not help.
Can someone answer the three questions in the previous post. Thanks.

aallan, 2023-04-12:
> There are there two differerent uf2 files in your repository.
> Whats the difference between picoprobe.uf2 and debugprobe.uf2?

The picoprobe.uf2 firmware is intended to be run on a Raspberry Pi Pico (see Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), the debugprobe.uf2 firmware is intended to be run on a Raspberry Pi Debug Probe (see https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html).

> Can you make a screenshot of the zadig program showing the correct windows 10 drivers? And which uf2 file is standard flashed in the debug probe?

Zadig is no longer required under any circumstances. See the two links above for official instructions.

NickDolfin, 2023-04-12:
Thanks,

I have changed the windows drivers using zadig. I want to restore them to the original, correct ones. Can you make a screenshot showing which drivers are correct?

w4mmp-ron, 2023-04-12:
Using the Device Manager, delete those devices that you used zadig to install drivers for.  When the device manager asks if you want to delete the driver also, click that check box.   remove the Pico or the debug probe, (I have lost track of what you have) and reboot the PC.  Now plug in the Pico or debug probe.  The device manager should show what I had previously posted.

NickDolfin, 2023-04-12:
Hi Ron,

I mentioned in the first sentence of the first post: 
I have a brand new Raspberry Pi Debug Probe (https://www.raspberrypi.com/products/debug-probe/)

I did what you wrote. This is what I get:
[image]

zooming in on one interfce:
[image]

you can seen that a libusbK driver is installed for interface1
and

[image]

for interface0 a libusb0 driver.

Windows does not have a default driver correctly installed after I changed things. It never will - thats windows. It remembers things based on the USB ID.

Can you show - just like I did - your installation settings with zadig.

w4mmp-ron, 2023-04-12:
Well, if you insist that zadig is required, I can't help.  It has been mentioned several times in this issue, that zadig can not be used with the the most recent firmware for the debug probe.  
The debug probe works fine on my Windows 10 PC.  I simply attach it to the Windows 10 PC and Windows installs the correct driver automatically.

NickDolfin, 2023-04-12:
I do not instist taht it is used. I just fucked up. So can you install zadig and show what your settings are?

w4mmp-ron, 2023-04-12:
Forget about zadig!   See the below.   If this is what you see when the probe is attached,  delete it along with the driver if the device manager offers that option.
[image]

w4mmp-ron, 2023-04-12:
Also, set the device manager to show hidden devices.  Delete all the hidden devices if they are using any of the zadig drivers.
[image]

NickDolfin, 2023-04-13:
For your and mine information:

Windows 10 is really a pain in the ass.
I ended up rstoring it from a backup that I made a few days ago (always a good idea to have backups).
The Raspberry Pi Debug Probe installed in an USB3 socket:

[image]
[image]
[image]

[image]
[image]
[image]

Thomas-Mcilwraith, 2024-07-18:
Hi there, 
I am currently having the same issue - is the previous post from [NickDolfin](https://github.com/NickDolfin) the correct configuration? that is exaclty what i have in my Zadig display (although newer versions now)