raspberrypi/debugprobe #82: why the newest FM can't detect the MCU?

MrsKan, 2023-05-01:
I clone the project and build the source code, got a .uf2 file, and I download it to the pico. when I use openocd to connect the picoprobe, it shows DAP init fail and the LED on the pico didn't light or shine. But the newest released .uf2 file can connected. any wrong did I made? It confused me.
[image]

P33M, 2023-05-02:
This is probably because I merged the debugprobe branch without reverting the board include to picoprobe. What happens if you swap the header file included here? https://github.com/raspberrypi/picoprobe/blob/master/src/picoprobe_config.h#L50

MrsKan, 2023-05-07:
thanks for update, it is great.

closed 2023-05-07 by MrsKan