raspberrypi/debugprobe #84: SWO/SWV output

WerWolv, 2023-05-12:
Hi!

I'm currently trying to get SWO / SWV output to work on my custom picoprobe PCB. I'm testing it all on a STM32 microcontroller through their IDE. Flashing and Debugging is working just fine, however I can't seem to get any printf output to work.

When using openOCD, it correctly recognizes that SWO has been enabled on the probe:
```
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: SWO-UART supported
Info : CMSIS-DAP: SWO-MANCHESTER supported
```

However, trying to debug with SWV enabled simply yields the message `Target does not support SWV`. I also tried pyOCD which launches fine but displays the warning `W SWV not initalized: Target does not have TPIU component [swv]`.

I'm certain that my microcontroller supports this feature as it can be used through a STLink debugger just fine.
Is there something more I need to do to get this to work or does picoprobe simply not support it? I saw some code in the codebase that suggested that it does. 

Thanks a lot for the great work!

P33M, 2023-05-12:
>When using openOCD, it correctly recognizes that SWO has been enabled on the probe:

SWO is neither advertised nor supported.
https://github.com/raspberrypi/picoprobe/blob/46eb924601651e7bbc79489e12a7839714ee8962/include/DAP_config.h#L102

Which version of openOCD are you using? Please post a complete terminal output.

WerWolv, 2023-05-12:
Sorry, I should probably have included that in the question. I simply assumed that it would be (at least partially) implemented already and I'm just being stupid :)

I did enable the manchester option in an attempt to get it to work. I also enabled the UART option and implemented a UART driver to get it to compile. 
I was mainly confused by the error message that popped up since it suggests that the probe just doesn't advertise itself as SWV capable before it even reached any of that code.

P33M, 2023-05-12:
If you are implementing SWO then you will need a PIO SM to do Manchester decode, an additional bulk IN endpoint for the data, and an RTOS thread to shift data from the PIO FIFO to the endpoint buffer. You also need to implement baudrate selection in response to the appropriate DAP command.

SWO support isn't in scope for Picoprobe, mainly because RP2040 doesn't support trace output.

closed 2023-05-12 by P33M