raspberrypi/debugprobe #85: How to compile for the debugprobe?

rewolff, 2023-06-15:
The pinout of the debugprobe is different than what's recomended for picoprobe on another pico. 

How is the official "debugprobe" firmware compiled? 

(I possibly have broken my debugprobe hardware. But loading my own project I have a "commandline" interface to set and clear GPIO pins and everything I tried seems to work. But flashing the original firmware binary again: No luck. To debug this further I tried running debugprobe firmware on the pico and couldn't get it to work. So I want to look at the sources to figure out what to do with the third pin that through resistors and buffers connects to one of the pins on the connector. )

rewolff, 2023-06-15:
Nope. Two debugprobes "broken" and I want to figure out what's wrong. 

If I can compile those myself, I can add debug-stuff, (e.g. output it on the virtual comport), or say change the pinout and patch the hardware to use a pin that DOES still work. 

Or assuming I have blown up the GPIO pads of the current DEBUG port...  I could patch some unused pins onto the port, and adjust the firmware. Or I could swap the uart and DBG ports (the hardware seems to be the same, right?) 

So my question remains: how do I compile that firmware myself?

P33M, 2023-06-15:
swap board_pico_config.h for board_debugprobe_config.h 
https://github.com/raspberrypi/picoprobe/blob/master/src/picoprobe_config.h#L51

closed 2023-06-15 by rewolff

rewolff, 2023-06-15:
Question: Would you accept a pull request with a short: how to compile these sources in "README.md"?

P33M, 2023-06-16:
Instructions for users are provided in the linked getting-started guide in the readme, but I wouldn't say no to a "hacking" section in readme.md