raspberrypi/debugprobe #88: Commit d04ff3b4724d93c2ffeb5fbe4f1d0665a36e7c29 is broken with probe-rs

whitequark, 2023-06-16:
Working: tag picoprobe-cmsis-v1.02
Broken: commit d04ff3b4724d93c2ffeb5fbe4f1d0665a36e7c29

To reproduce, run `probe-rs-cli info` twice with any target connected; the first run succeeds, the second run fails with this log:
```
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: CMSIS-DAP v2 Interface
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 2, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Maya's Debug Probe: CMSIS-DAP device with 3 interfaces
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using libusb, searching HID
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_all:list_jlink_devices: jaylink: libusb 1.0.24.11584    
DEBUG list_all:list_jlink_devices: jaylink: libusb has capability API: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb has HID access: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb has hotplug support: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb can detach kernel driver: true    
TRACE open: probe_rs::probe::cmsisdap::tools: Attempting to open device matching 2e8a:000c:E6614C309349912A
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 004 Device 001: ID 1d6b:0003
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 003 Device 001: ID 1d6b:0002
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 001: ID 1d6b:0003
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 003: ID 27c6:5395
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 013: ID 8087:0025
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 004: ID 0c45:6d13
TRACE open: probe_rs::probe::cmsisdap::tools: Error getting languages: Timeout
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 025: ID 2e8a:000c
TRACE open: probe_rs::probe::cmsisdap::tools:   Interface 0: CMSIS-DAP v2 Interface
TRACE open: probe_rs::probe::cmsisdap::tools: Could not read string for interface 2, skipping
TRACE open: probe_rs::probe::cmsisdap::tools: Maya's Debug Probe: CMSIS-DAP device with 3 interfaces
TRACE open: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG open: probe_rs::probe::cmsisdap::tools: Opening 2e8a:000c in CMSIS-DAPv2 mode
DEBUG open: probe_rs::probe::cmsisdap::commands: Draining probe of any pending data.
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 1 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 2 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 3 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 4 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 5 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 6 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 7 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 8 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 9 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 10 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 11 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 12 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 13 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 14 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 15 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 16 to find packet size
TRACE open: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
```

Wren6991, 2023-06-17:
Thanks, confirmed. probe-rs also seems to put the probe into a state where openocd can no longer attach, citing:

```
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=D769B0835B6D3E21
Error: CMSIS-DAP command CMD_INFO failed.
```

So something fundamental is broken. I'm travelling at the moment so my debug setup is a bit hoopty, but I can start looking properly on Monday.

Wren6991, 2023-06-17:
For me `probe-rs-cli info` doesn't seem to get very far at all. On a fresh start I get 3x `DAP_Info`, then a `DAP_Disconnect`, then it hangs. Meanwhile OpenOCD starts up with a couple of `DAP_Info` followed by a  `DAP_Connect` before setting up pins, clocks etc. I wonder if we're just failing to handle a disconnect that's not preceded by a matching connect.

Wren6991, 2023-06-17:
Ah, yeah, that would do it:

```patch
diff --git a/src/probe.c b/src/probe.c
index 70e64b1..0abf9ca 100644
--- a/src/probe.c
+++ b/src/probe.c
@@ -157,8 +157,8 @@ void probe_init() {
 
 void probe_deinit(void)
 {
-  probe_read_mode();
   if (probe.initted) {
+    probe_read_mode();
     pio_sm_set_enabled(pio0, PROBE_SM, 0);
     pio_remove_program(pio0, &probe_program, probe.offset);
     probe.initted = 0;
```

commit 0746b5a8 (2023-06-17): Don't call probe_read_mode() in deinit() without matching prior init(), fixes #88
commit 7de418cc (2023-06-18): Merge pull request #89 from raspberrypi/fix-deinit-without-init

Don't call probe_read_mode() in deinit() without matching prior init(), fixes #88
closed 2023-06-18 by P33M