raspberrypi/debugprobe #92: Readme doesn't mention pico-sdk, so picoprobe can't be compiled

cdwijs, 2023-08-12:
On an arch linux machine without pico-sdk, the compile fails. Readme.md doesn't mention the need for pico-sdk to be installed.
First remove pico-sdk to emulate a system where it was never installed:
```
# pacman -R pico-sdk
```
Reproduce the error:
```
$ git clone https://github.com/raspberrypi/picoprobe.git
$ cd picoprobe/
$ git submodule update --init
$ mkdir build
$  cd build
$ cmake ..
Using PICO_SDK_PATH from environment ('/usr/share/pico-sdk')
CMake Error at pico_sdk_import.cmake:63 (message):
  Directory '/usr/share/pico-sdk' not found
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)
-- Configuring incomplete, errors occurred!
```

This can be solved on arch linux like this:
```
# pacman -S arm-none-eabi-binutils arm-none-eabi-gcc arm-none-eabi-newlib
$ wget https://aur.archlinux.org/cgit/aur.git/snapshot/pico-sdk.tar.gz
$ tar -xf pico-sdk.tar.gz
$ cd pico-sdk
$ makepkg
# pacman -U pico-sdk-1.5.1-1-any.pkg.tar
```
Then the cmake step works:
```
$ cmake ..
$ make
$ ls -l picoprobe.uf2 
-rw-r--r-- 1 cedric cedric 80896 Aug 12 11:48 picoprobe.uf2
```

cdwijs, 2023-08-12:
I've made a pull request that addresses this issue:
https://github.com/raspberrypi/picoprobe/pull/93

lurch, 2023-08-17:
`Using PICO_SDK_PATH from environment ('/usr/share/pico-sdk')` - that looks like something that you must have set on your system? All of our official instructions say to clone pico-sdk from GitHub, rather than to install it to some system-location like `/usr/share/pico-sdk`.
If you **didn't** have a `PICO_SDK_PATH` envvar set, then you'd get an error like:
```console
$ cmake ..
CMake Error at pico_sdk_import.cmake:55 (message):
  SDK location was not specified.  Please set PICO_SDK_PATH or set
  PICO_SDK_FETCH_FROM_GIT to on to fetch from git.
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!
```

You're right that the README.md here doesn't mention cloning the pico-sdk in the [Hacking](https://github.com/raspberrypi/picoprobe/blob/master/README.md#hacking) section (perhaps it should), but the PDF documentation pointed to in the [Documentation](https://github.com/raspberrypi/picoprobe/blob/master/README.md#documentation) section is correct.

cdwijs, 2023-08-20:
Thank you. I've successfully built picoprobe by first cloning the SDK,and then cloning the picoprobe repository. I think cloning all the repositories, and then building the target program is a cleaner way than installing pico-sdk system wide. I'll change my pull request to reflect this.

[code]
$ cd ~/
$ mkdir pico
$ cd pico

$ git clone https://github.com/raspberrypi/pico-sdk.git --branch master
$ cd pico-sdk
$ git submodule update --init
$ cd ..
$ git clone https://github.com/raspberrypi/pico-examples.git --branch master #optional

$ git clone https://github.com/raspberrypi/picoprobe.git
$ cd picoprobe
$ git submodule update --init
$ mkdir build
$ cd build
$ export PICO_SDK_PATH=../../pico-sdk
$ cmake ..
$ make -j4
[/code]

Complete output excluding the build itself:
[code]
[cedric@cedric ~]$ cd
[cedric@cedric ~]$ mkdir pico
[cedric@cedric ~]$ cd pico
[cedric@cedric pico]$ git clone https://github.com/raspberrypi/pico-sdk.git --branch master
Cloning into 'pico-sdk'...
remote: Enumerating objects: 7587, done.
remote: Counting objects: 100% (2242/2242), done.
remote: Compressing objects: 100% (410/410), done.
remote: Total 7587 (delta 1889), reused 1890 (delta 1814), pack-reused 5345
Receiving objects: 100% (7587/7587), 2.61 MiB | 3.49 MiB/s, done.
Resolving deltas: 100% (4163/4163), done.
[cedric@cedric pico]$ cd pico-sdk
[cedric@cedric pico-sdk]$ git submodule update --init
Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
Submodule 'lib/cyw43-driver' (https://github.com/georgerobotics/cyw43-driver.git) registered for path 'lib/cyw43-driver'
Submodule 'lib/lwip' (https://github.com/lwip-tcpip/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mbedtls' (https://github.com/Mbed-TLS/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'tinyusb' (https://github.com/hathach/tinyusb.git) registered for path 'lib/tinyusb'
Cloning into '/home/cedric/pico/pico-sdk/lib/btstack'...
Cloning into '/home/cedric/pico/pico-sdk/lib/cyw43-driver'...
Cloning into '/home/cedric/pico/pico-sdk/lib/lwip'...
Cloning into '/home/cedric/pico/pico-sdk/lib/mbedtls'...
Cloning into '/home/cedric/pico/pico-sdk/lib/tinyusb'...
Submodule path 'lib/btstack': checked out '72ef1732c954d938091467961e41f4aa9b976b34'
Submodule path 'lib/cyw43-driver': checked out '8ef38a6d32c54f850bff8f189bdca19ded33792a'
Submodule path 'lib/lwip': checked out '239918ccc173cb2c2a62f41a40fd893f57faf1d6'
Submodule path 'lib/mbedtls': checked out 'a77287f8fa6b76f74984121fdafc8563147435c8'
Submodule path 'lib/tinyusb': checked out '86c416d4c0fb38432460b3e11b08b9de76941bf5'
[cedric@cedric pico-sdk]$ cd ..
[cedric@cedric pico]$ git clone https://github.com/raspberrypi/pico-examples.git --branch master
Cloning into 'pico-examples'...
remote: Enumerating objects: 2763, done.
remote: Counting objects: 100% (148/148), done.
remote: Compressing objects: 100% (97/97), done.
remote: Total 2763 (delta 64), reused 118 (delta 50), pack-reused 2615
Receiving objects: 100% (2763/2763), 7.93 MiB | 3.13 MiB/s, done.
Resolving deltas: 100% (1394/1394), done.
[cedric@cedric pico]$ pwd
/home/cedric/pico
[cedric@cedric pico]$ git clone https://github.com/raspberrypi/picoprobe.git
Cloning into 'picoprobe'...
remote: Enumerating objects: 357, done.
remote: Counting objects: 100% (357/357), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 357 (delta 257), reused 241 (delta 191), pack-reused 0
[... 50 lines cut ...]

After the build picoprobe.uf2 has been made
[code]
[cedric@cedric build]$ ls -l
total 1628
-rw-r--r-- 1 cedric cedric  20368 Aug 20 18:55 CMakeCache.txt
-rw-r--r-- 1 cedric cedric  14828 Aug 20 18:55 CMakeDoxyfile.in
-rw-r--r-- 1 cedric cedric  20469 Aug 20 18:55 CMakeDoxygenDefaults.cmake
drwxr-xr-x 1 cedric cedric    562 Aug 20 18:58 CMakeFiles
-rw-r--r-- 1 cedric cedric   1767 Aug 20 18:55 cmake_install.cmake
drwxr-xr-x 1 cedric cedric    160 Aug 20 18:58 elf2uf2
drwxr-xr-x 1 cedric cedric     74 Aug 20 18:55 FREERTOS_KERNEL
drwxr-xr-x 1 cedric cedric     18 Aug 20 18:55 generated
-rw-r--r-- 1 cedric cedric 148789 Aug 20 18:55 Makefile
-rwxr-xr-x 1 cedric cedric  40500 Aug 20 18:58 picoprobe.bin
-rw-r--r-- 1 cedric cedric 764472 Aug 20 18:58 picoprobe.dis
-rwxr-xr-x 1 cedric cedric 101908 Aug 20 18:58 picoprobe.elf
-rw-r--r-- 1 cedric cedric 339757 Aug 20 18:58 picoprobe.elf.map
-rw-r--r-- 1 cedric cedric 113980 Aug 20 18:58 picoprobe.hex
-rw-r--r-- 1 cedric cedric  81408 Aug 20 18:58 picoprobe.uf2
drwxr-xr-x 1 cedric cedric     98 Aug 20 18:55 pico-sdk
drwxr-xr-x 1 cedric cedric    114 Aug 20 18:58 pioasm
-rw-r--r-- 1 cedric cedric   3157 Aug 20 18:58 probe_oen.pio.h
-rw-r--r-- 1 cedric cedric   2926 Aug 20 18:58 probe.pio.h
[/code]

aallan, 2023-08-21:
Closing, see my comment on #93.

closed 2023-08-21 by aallan