raspberrypi/debugprobe #95: [FEATURE REQUEST] make target for PROTO_DAP_V1

xxxajk, 2023-08-15:
Would be nice to have a make target that defines ```PICOPROBE_DEBUG_PROTOCOL``` to  ```PROTO_DAP_V1```.
This makes it compatible with older OpenOCD (and other programs that are closed source) which wants to communicate via HID instead of bulk.
Bonus points for being able to do both protocols, based on which pipe the data comes in on to change between the two versions.

P33M, 2023-08-15:
The DAP_V1 defines are included for posterity and will not be actively supported. The minimum guarantee is that defining PROTO_DAP_V1 won't break compilation. There won't be a release that targets HID as a transport protocol - I would encourage your ossified vendors to bring their software stacks into the 21st century, which is nearly one quarter gone already.

closed 2023-08-15 by P33M

xxxajk, 2023-08-16:
Hard to do when it is embedded :-/

lurch, 2023-08-16:
Debug protocols are something I know nothing about, but I **do** remember seeing discussion of various picoprobe-alternatives, so maybe one of those will do what you're looking for?
#2 
#78

xxxajk, 2023-08-17:
> Debug protocols are something I know nothing about, but I **do** remember seeing discussion of various picoprobe-alternatives, so maybe one of those will do what you're looking for? #2 #78

Well, I could take what's here, and "make it my own", which is what I was going to do and have to do anyway.
Upstream added SRST, which is required for any target that is non-pi basically.  
I will probably add more features and should easily be able to expand to full JTAG. 
I have a plain non-swd tool I made myself a long while ago, based on other tools supported under OpenOCD, but the PICO is a great option for the price, and can run faster than my old implementation.
Furthermore I know of a few speed enhancements that can increase the JTAG (and probably SWD) data rates while running at the same basic clock rate (sort of).