raspberrypi/debugprobe #98: Support for parity

kirkegaard, 2023-09-18:
Would it be possible to add parity support to pico probe? Im unsure what it entails but i was told it was missing from the probe.

Regarding this request: https://github.com/grigorig/stcgal/issues/94

P33M, 2023-09-18:
Yes, but the pico-sdk API doesn't have a neat way of setting all the line coding parameters at once - uart_init only takes baudrate as a parameter and the functions of interest (e.g. uart_set_config) are statically defined. A workaround would be to open-code the register writes.

P33M, 2023-09-18:
Derp, uart_set_format is indeed useable.

Does this branch work for you: https://github.com/raspberrypi/picoprobe/tree/line_coding

kirkegaard, 2023-09-18:
YES! It works perfectly! Thank you so much :)

closed 2023-09-18 by kirkegaard