raspberrypi/debugprobe #104: Picoprobe/OpenOCD can crash, requiring power cycle

byteit101, 2023-10-23:
Unsure where this should be reported, but about once every 5 minutes I have to pull the power on the picoprobe and DUT and plug them both in again to reset them. During this time, Serial/UART bridge is broken and doesn't work too. I'm not sure what triggers this, as it usually spontaneously happens when I'm not doing anything in particular beyond UART reading. Sometimes I'm connected via GDB, sometimes I'm not. If I only reset power for the picoprobe, the led stays off until I reset power to the DUT too.

Symptoms via OpenOCD:

```
...
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                                                                                 
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_SWD_Configure failed.                                                                                           
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_DISCONNECT failed.                                                                                              
Error: error writing data: No such device (it may have been disconnected)                                                                    
Error: CMSIS-DAP command CMD_CONNECT failed.                                    
...
```

P33M, 2024-11-07:
I have sort-of reproduced this. If you connect the probe to an unpowered Pi, then jiggle the Pi power connector in-out you can get the probe to stop responding (but UART LEDs still flash, so it's not crashed).

P33M, 2024-11-20:
This is probably a manifestation of RP2040-E15. If you have floating grounds between boards without a ground connection, then ESD protection circuits at either end of the USB connection can activate and this has the effect of disrupting Dp/Dm. The same also applies with poor quality cables.

The device controller is vulnerable to certain linestate errors and can lock up. The only way to get out of this state is to force a disconnect and reset the peripheral. I implemented it on a branch here - https://github.com/raspberrypi/debugprobe/tree/watchdog

closed 2025-11-24 by P33M