raspberrypi/debugprobe #105: Multi-device support?

DatanoiseTV, 2023-10-25:
Is there any chance to add a multi device programming option? I am currently designing a board
with 4 RP2040 on it and just placing another rp2040 on board as picoprobe and as debug interface would be great.

JustAnother1, 2023-11-17:
I don't think that that would be something that picoprobe needs to deal with.

You probably have to change the instance is of the 4 RP2040 to avoid collisions. Default ids are: 
Core 0: 0x01002927
Core 1: 0x11002927
Rescue DP: 0xf1002927

But these can also be used:
 0x21002927, 0x31002927, 0x41002927, 0x51002927, 0x61002927, 0x71002927
 0x81002927, 0x91002927, 0xa1002927, 0xb1002927, 0xc1002927, 0xd1002927,
 0xe1002927

Details on how to change them is in the RP2040 datasheet.

Then it is only an issue with OpenOCD or whatever GDBServer you are using to use the correct id.

Alternatively you can "somehow" switch the SWDIO Line to the chip you want to debug.

P33M, 2024-11-07:
RP2040 supports multidrop SWD, so no changes needed to the probe firmware - you need a platform-specific method of keeping all but the first RP2040 in reset, then programming the DP ID for each RP2040 in turn. It'll be a lot of OpenOCD scripting.

closed 2024-11-07 by P33M