raspberrypi/debugprobe #107: Don't know where the problem exists, picoprobe firmware, openocd version, some missing udev rules, something entirely else.

dx9s, 2023-11-13:
I cannot get openocd to see the picoprobe, but it appears to be there. HOPEFULLY this is enough information for others to chime in:
```
dx@pop-os:~/pico/picoprobe/build$ git status
HEAD detached at picoprobe-cmsis-v1.0.3
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	./

nothing added to commit but untracked files present (use "git add" to track)
dx@pop-os:~/pico/picoprobe/build$ make
[  2%] Built target bs2_default
[  4%] Built target bs2_default_padded_checksummed_asm
[  5%] Performing build step for 'PioasmBuild'
Consolidate compiler generated dependencies of target pioasm
[100%] Built target pioasm
[  6%] No install step for 'PioasmBuild'
[  7%] Completed 'PioasmBuild'
[ 11%] Built target PioasmBuild
[ 12%] Built target picoprobe_probe_pio_h
[ 13%] Built target picoprobe_probe_oen_pio_h
[ 14%] Performing build step for 'ELF2UF2Build'
Consolidate compiler generated dependencies of target elf2uf2
[100%] Built target elf2uf2
[ 15%] No install step for 'ELF2UF2Build'
[ 15%] Completed 'ELF2UF2Build'
[ 19%] Built target ELF2UF2Build
Consolidate compiler generated dependencies of target picoprobe
[100%] Built target picoprobe
dx@pop-os:~/pico/picoprobe/build$ cp *.uf2 /media/dx/RPI-RP2/.
dx@pop-os:~/pico/picoprobe/build$ lsusb | grep -i pi
Bus 009 Device 047: ID 2e8a:000c Raspberry Pi Picoprobe (CMSIS-DAP)
dx@pop-os:~/pico/picoprobe/build$ openocd -f interface/cmsis-dap.cfg -f interface/picoprobe.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-11-12-20:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
adapter speed: 5000 kHz

adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to find a matching CMSIS-DAP device

dx@pop-os:~/pico/picoprobe/build$ ls -la /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 Nov 12 20:26 /dev/ttyACM0
dx@pop-os:~/pico/picoprobe/build$ sudo dmesg |tail -20
[sudo] password for dx: 
[127466.813410] usb-storage 9-1.1:1.0: USB Mass Storage device detected
[127466.813609] scsi host6: usb-storage 9-1.1:1.0
[127467.838855] scsi 6:0:0:0: Direct-Access     RPI      RP2              2    PQ: 0 ANSI: 2
[127467.839055] sd 6:0:0:0: Attached scsi generic sg4 type 0
[127467.839802] sd 6:0:0:0: [sdd] 262144 512-byte logical blocks: (134 MB/128 MiB)
[127467.841264] sd 6:0:0:0: [sdd] Write Protect is off
[127467.841266] sd 6:0:0:0: [sdd] Mode Sense: 03 00 00 00
[127467.842906] sd 6:0:0:0: [sdd] No Caching mode page found
[127467.842910] sd 6:0:0:0: [sdd] Assuming drive cache: write through
[127467.851604]  sdd: sdd1
[127467.851987] sd 6:0:0:0: [sdd] Attached SCSI removable disk
[127479.368391] usb 9-1.1: USB disconnect, device number 46
[127479.376902] FAT-fs (sdd1): unable to read boot sector to mark fs as dirty
[127479.725736] usb 9-1.1: new full-speed USB device number 47 using xhci_hcd
[127479.836575] usb 9-1.1: New USB device found, idVendor=2e8a, idProduct=000c, bcdDevice= 1.03
[127479.836579] usb 9-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[127479.836581] usb 9-1.1: Product: Picoprobe (CMSIS-DAP)
[127479.836582] usb 9-1.1: Manufacturer: Raspberry Pi
[127479.836584] usb 9-1.1: SerialNumber: E66098F29B8B4E32
[127479.852673] cdc_acm 9-1.1:1.1: ttyACM0: USB ACM device
dx@pop-os:~/pico/picoprobe/build$ id
uid=1000(dx) gid=1000(dx) groups=1000(dx),4(adm),20(dialout),22(voice),24(cdrom),27(sudo),29(audio),123(lpadmin),137(libvirt)
```

I don't know WHY openocd (which is built from the code preference in the "Getting started with Paspberry Pi Pico" Appendix A: Using Picoprobe.

The instructions seem quite straight forward but I am missing something.

--Doug (dx9s)

FWIW, my user account is in the dialout group which grants access to the serial port /dev/ttyACM0

dx9s, 2023-11-13:
FWIW.. I've tried with -f ` interface/picoprobe.cfg ` alone and with -f ` interface/cmsis-dap.cfg ` alone as I've seen both used in different places.

dx9s, 2023-11-13:
Okay... tried it with sudo:
```
dx@pop-os:~$ sudo openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
[sudo] password for dx: 
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-11-12-20:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E66098F29B8B4E32
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : DAP init failed
```
(the pico under debug is not connected, but this is further that previous)

It didn't occur to me that it is some other permission (outside serial) that is being blocked. Hopefully there is a udev rule that should work.

--Doug

dx9s, 2023-11-13:
Okay found a Stack Exchange article (Arduino related) that was close enough to create a udev rule:
```
dx@pop-os:/etc/udev/rules.d$ cat 98-openocd.rules 
ACTION!="add|change", GOTO="openocd_rules_end"
SUBSYSTEM!="usb|tty|hidraw", GOTO="openocd_rules_end"
ATTRS{product}=="*CMSIS-DAP*", MODE="666", GROUP="plugdev"
LABEL="openocd_rules_end" 
```
performed a ` sudo udevadm control --reload-rules `

and unplugged and replugged in my picoprobe and the following works as a normal user:
```
dx@pop-os:~$ openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-11-12-20:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E66098F29B8B4E32
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : DAP init failed
```

Might want to update the openocd source code to include a udev rule that is installed via the ` sudo make install `

P33M, 2024-11-07:
Udev isn't guaranteed to exist on the target system. Also there's an example file in contrib/60-openocd.rules. In any case, this is out of scope for this repo.

closed 2024-11-07 by P33M