raspberrypi/debugprobe #112: Request 2 cancelled on connection close

victoryred, 2023-12-08:
Hi All,
The problem I'm having getting picoprobe debug started relates to the title of this issue ticket.

I'm using the Arduino-ide v-2.2.1 along with a Raspberry Pi pico setup as an swd for a Raspberry Pi pico target.

Debugging halts but it also says there are 4 breakpoints, and 2 watchpoints. Those break/watchpoints are remenants from a previous session (persistant storage? Where?).

/home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd" -f "interface/cmsis-dap.cfg" -f "target/rp2040.cfg" -s "/home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/share/openocd/scripts" -c "adapter speed 5000" -c "program {/tmp/arduino/sketches/E5E0EC72C59786124DE369A39273D71F/GetCycleCount.ino.elf} verify" -c "reset init" -c "resume" -c "exit"
Open On-Chip Debugger 0.12.0-g4d87f6dca (2023-09-11-23:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E661640843626C27
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: ### 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00**
** Programming Started **
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : Padding image section 1 at 0x100125bc with 68 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10012600 .. 0x1001ffff
** Programming Finished **
** Verify Started **
** Verified OK **
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00

lurch, 2023-12-09:
> Debugging halts but it also says there are 4 breakpoints, and 2 watchpoints. Those break/watchpoints are remenants from a previous session (persistant storage? Where?).

That's possibly a question for the developers of the "Arduino-ide v-2.2.1" that you're using? :shrug:

victoryred, 2023-12-10:
Yes, my thoughts, too. BTW, I'm running on a Debian X86 machine. The more interesting question however is this. In the OUTPUT window of the IDE, the Info: messages are generated by different parts of the suite. For example, some info messages tell you explicitly that the trace message is in Openocd, other messages are from cmsis-dap. But the messages beginning with Info : [rp2040.core0] look like they are from SWD. I think that's where the problem is. At this point I'm not the guy to fix it without guidance.

On another note- notifications in the Arduino-IDE are simply posts from the originating program. When the messages are terse, they aren't helpful. The issuing program name should precede the message.

[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00

lurch, 2023-12-10:
> At this point I'm not the guy to fix it without guidance.

Raspberry Pi don't provide official support for the Arduino environment, so you probably want to ask for further help at https://github.com/earlephilhower/arduino-pico/issues and/or https://forums.raspberrypi.com/viewforum.php?f=143

victoryred, 2023-12-14:
Thanks for the pointer. 
I reached out to E.P. and got a redirect to Raspberrypi since they are maintaining picoprobe for their device. FWIW, I've put in a big effort to troubleshoot in a command line environment. I started openocd in terminal 1 with -f to the interfaces and targets. Went well. All cfg files are Tcl scripts. Opened gdb-multiarch sha1.ino in terminal 2. I opened localhost:3333, connected and loaded. I typed monitor reset init and got the same halt. The Openocd document agrees that reset init should work that way. But in the arduino-ide v2.2.1 the connection to the second port (on the target pico clk & data debug port) closes- no further communication.

closed 2024-11-07 by P33M