raspberrypi/debugprobe #118: Feature suggestion: add ability to unlock flash sectors

dmo9, 2024-02-06:
Some of my code requires the XIP system to be disabled, and when testing this, I accidentally protected all sectors of the flash chip. This means that I cannot update any of my code via the probe and had to resort to using external tools to do so.

It would be nice if the probe automatically unlocked sectors.

P33M, 2024-02-06:
That's not the job of a debug adapter that merely bridges communication from a controlling application (openocd) to a debug target (rp2040, or any other SWDv2 DAP).

closed 2024-02-06 by P33M

dmo9, 2024-02-09:
For anyone wondering how I fixed the issue: I desoldered the flash chip & then used an adapter + genuine Xgecu TL866II+ to reprogram status register 1. Three other alternatives that I also think would work:
1. A second microcontroller with SPI to send the commands to status register 1.
2. An FTDI chip / cable that acts as a USB to SPI bridge.
3. A Segger debug probe (must use version higher than their base model) + JFlash.

lurch, 2024-02-09:
> A second microcontroller with SPI to send the commands to status register 1.

Couldn't you use a load-to-RAM UF2 file to do that? (i.e. without having to desolder your flash chip)
A similar approach to what is done by https://github.com/raspberrypi/pico-examples/tree/master/flash/nuke

dmo9, 2024-02-09:
> > A second microcontroller with SPI to send the commands to status register 1.
> 
> 
> 
> Couldn't you use a load-to-RAM UF2 file to do that? (i.e. without having to desolder your flash chip)
> 
> A similar approach to what is done by https://github.com/raspberrypi/pico-examples/tree/master/flash/nuke

Good call, didn't even think about that; I have a nice soldering setup so I tend to lean to that maybe more than I should.