raspberrypi/debugprobe #119: Probe can no longer connect to Pico after rename

Swiftloke, 2024-02-08:
Looks like the rename broke things. Ran into this while trying to set up my Pico as per Appendix A of the documentation. The last working commit was 1267a8c367c73ef659cac3fde22ffc3e3074fd99. After that commit, the code is broken— The LED on the board no longer lights up, and while OpenOCD can connect to it, it bails out with the error:

```
Error: Failed to connect multidrop rp2040.dap0
```

P33M, 2024-02-09:
Which UF2 have you loaded on your Pico?

Swiftloke, 2024-02-10:
the ``debugprobe.uf2`` file generated by ``make``, as per the build instructions

P33M, 2024-02-12:
The build instructions in https://github.com/raspberrypi/debugprobe?tab=readme-ov-file#hacking say to use `debugprobe_on_pico.uf2`. Use these until the documentation release is updated.

Swiftloke, 2024-02-12:
Ah, my apologies, I thought you asked what file I had loaded onto my debug probe Pico. I am using a two-Pico setup, with the debug Pico connecting to the target Pico.

I loaded ``debugprobe.uf2`` onto the Pico I intend to use as a debug adapter. The target Pico just has the blinking LED program from the examples.

Note that in the commits before the rename, loading ``picoprobe.uf2`` onto the debug adapter Pico allows it to connect to the target Pico without issue. That is what's broken here.

Swiftloke, 2024-02-15:
I can confirm that that version of the firmware works. Since this is an issue surrounding a confusing name and documentation, I'm closing this issue and moving over to the pico-feedback repository, as seen above.

closed 2024-02-15 by Swiftloke