raspberrypi/debugprobe #120: How to use target reset functionality?

recursivenomad, 2024-02-09:
I am interested in having the option to hard-reset my target Pico and its peripherals by pulling its 3V3_EN pin low.  I've built picoprobe by changing:

```c
#if false
#define PROBE_PIN_RESET 1
#endif
```

to:

```c
#if true
#define PROBE_PIN_RESET 1
#endif
```

in [board_pico_config.h](https://github.com/raspberrypi/debugprobe/blob/picoprobe-cmsis-v1.1/include/board_pico_config.h).  However, when I pass `monitor reset init` to GDB, it appears to be restarting as normal.  I tried `monitor reset_config srst_only` and `monitor adapter srst pulse_width 1000` to try and visibly see a reset happen, but it simply delays the ability to enter a command in GDB, never powering-down the target Pico.  It looks like [the PR that introduced the feature](https://github.com/raspberrypi/debugprobe/pull/10) mentions "This can easily be added to OpenOCD", but I'm struggling to understand how.  I'm new to using all these tools so I'm certainly taking shots in the dark with these efforts.

Before I keep trying, is there a known way to drive PROBE_PIN_RESET low through OpenOCD via a GDB `monitor` command, or otherwise?

lurch, 2024-02-13:
I _believe_ that the intention is that you'd wire PROBE_PIN_RESET up to the [RUN pin](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#pinout-and-design-files) on the target board that you're debugging? I.e. that's why it's a "hardware reset" rather than a "software reset".

(Ahh, although re-reading your original post, perhaps I've misunderstood what it is that you're asking?)

recursivenomad, 2024-02-13:
Correct, I've been experimenting with wiring to both the RUN pin and the 3V3_EN pin.  My trouble is understanding how to use this functionality once recompiling and uploading the modified picoprobe firmware - It doesn't appear to be hardware-resetting the target Pico (although I'll admit I haven't scoped it with a logic analyzer yet).

I'm curious if there's a specific command I can send through GDB/OpenOCD to manually activate the reset line functionality?  Looking for documentation on how to use this feature after it's already been wired up.

lurch, 2024-02-13:
This is just a guess, but I wonder if that might be part of the old "custom picoprobe interface", and not yet part of the (now-default) "standard cmsis-dap interface" ? :shrug: 

Ahh... looks like https://github.com/raspberrypi/debugprobe/commit/076142482157ae91f4f459f03253e7a45c36825f is the commit that hooks it into the "standard cmsis-dap interface", so is probably a more-useful starting point than #10 ?

P33M, 2024-02-13:
Can you try the options listed in the OpenOCD user guide? https://openocd.org/doc/html/Reset-Configuration.html

recursivenomad, 2024-02-20:
Okay, finally found the time to test everything with a logic analyzer attached 🙂  Bottom line, none of the configuration options I tried from that OpenOCD page managed to produce a low signal on PROBE_PIN_RESET (I was monitoring the probe's pins 0 and 1 for good measure).

That said, I *can* confirm that sending `monitor cmsis-dap info` through GDB returns `... nRESET = 1`, which is a promising start.

Adding `"reset_config srst_only"` to `"openOCDLaunchCommands"` in my lauch.json changes the output of `monitor reset_config`, but still no low signal on PROBE_PIN_RESET.

And same goes for manually sending `monitor adapter assert srst` via GDB.  No signal.

Setting `monitor debug 3` and viewing the OpenOCD output shows that OpenOCD believes SRST is asserted during reset; but again, no change in signal.

I looked into the target configuration file and did find this in OpenOCD's `.../target/rp2040.cfg`:

```cfg
	# srst does not exist; use SYSRESETREQ to perform a soft reset
	$_TARGETNAME_0 cortex_m reset_config sysresetreq
```

However, I don't believe this is related as OpenOCD believes it is asserting and releasing SRST during reset when it is enabled.

P33M, 2024-03-18:
Fixed in #123

closed 2024-03-18 by P33M