raspberrypi/debugprobe #121: Fatal error: bsp/board.h: No such file or directory

richard-blazek, 2024-02-22:
When compiling debugprobe, I get the following error: "fatal error: bsp/board.h: No such file or directory." In the file `main.c`, there are these two includes:

```c
#include "bsp/board.h"
#include "tusb.h"
```

But the files `tusb.h.` and `bsp/board.h` are not in the project. Where are they and how can I make it compile?

zouming74, 2024-03-25:
Can you share your experience about how you fix it ?

zouming74, 2024-03-25:
I solved this bug ,just run `git submodule update --init` in pico-sdk .

Slion, 2024-10-07:
At some point `board.h` was renamed to `board_api.h` it seems.
Just do:
```C
#include "bsp/board_api.h"
```

lurch, 2024-10-07:
> At some point `board.h` was renamed to `board_api.h` it seems.

Looks like it was renamed in August 2023 https://github.com/hathach/tinyusb/commit/cf91660cee0863291fd9412a3623df018e8bd071

@Slion Would you like to create a suitable PR?

lurch, 2024-10-07:
Oh, but see https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/tinyusb/include/bsp/board.h

Are you using an old version of pico-sdk @Slion ?

Slion, 2024-10-07:
Yeah I'm still on 1.5.1 with some updated submodules, including a custom branch of TinyUSB. So basically my own branch of pico_sdk. They renamed it twice then, in effect 😁
Oh well I'll change it back if need be when I eventually switch to 2.x SDK.
No PR need, I'll deal with that in my code. Thanks for the pointers though.

closed 2024-10-07 by lurch