raspberrypi/debugprobe #124: Unable to use debugprobe

emanuellopes, 2024-02-29:
Hello,

I'm testing the debugprobe (hardware tool) in my mac os.

I've installed all the binaries like documentation describes.

After doing everything I did the following command in console

```bash
$ sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"
```

But I received this:
```bash
Open On-Chip Debugger 0.12.0-g4d87f6d (2024-02-16-16:16)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6616407E39A712C
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Error: Failed to connect multidrop rp2040.dap0
```

I checked everything on pico W:
- cables
- Power
- Different power supply
- Power pico over vbus and vsys

Nothing happened, I'm always with the same issue.

I've used the latest debugprobe.uf2 on releases.

kamuycikap9821, 2024-03-03:
Hello.

I believe the OpenOCD you have created is complete.
I believe that the error is a correct result.

I would expect that if you set up PicoProbe, connect it to your Macbook, and then start it, you would not get the error.

A blog I wrote might be helpful.
https://kamuycikap.hatenablog.com/entry/2024/03/02/201039
*It is in Japanese, but I think you can read the command prompt.

Good luck.

emanuellopes, 2024-03-03:
Some days ago I was able to debug using debugprobe, the first time doesn't work and I notice the cables are not connected correctly.

After that the debugging started and I was able to get data in serial console.

After that I don't know what happened but the debugprobe didn't work anymore.

I have read your post and I did everything you have did.

Should I connect debugprobe after all connections are done in pico? Today I will test it again.

kamuycikap9821, 2024-03-04:
Hello.

Yes, I am looking for a new job.
Run the command with the PC, PicoProbe and target pico connected.

I also compiled and installed "debugprobe" but it did not work correctly.
I used picoprobe.uf2 which is released on GitHub instead of debugprobe.
https://github.com/raspberrypi/debugprobe/releases

One thing I am a little concerned about is that the terminal where OpenOCD is running is different from the terminal where GDB is running.

After running OpenOCD first, I leave that behind and open a new terminal to start GDB.

If your computer, PicoProbe, and target Pico are all connected, it should work.

P33M, 2024-11-07:
This is user error and not a bug. If you still have issues, please post on the Pico section of our forums.

closed 2024-11-07 by P33M

jonesalk, 2025-05-05:
I know you said that you cheeked the cables, but this is exactly the error message I see, 
if I  swap SWDIO and SWDCLK by mistake or if the my crusty DuPont cables fall out from breadboard (using spare pico board as probe).