raspberrypi/debugprobe #127: UART doesn't support transmitting a break signal

HiassofT, 2024-03-16:
Currently the CDC ACM UART of the debug probe doesn't support transmitting a break signal. eg press `CTRL-A` `F` in minicom and observe that nothing happens.

Having support for break transmission would be very handy, eg to activate SYSRQ in Linux, get into kgdb/U-Boot etc

The Linux CDC ACM driver has support for break transmission https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/usb/class/cdc-acm.c#L914-L927 if the usb device supports it (USB_CDC_CAP_BRK / 0x04 in CDC ACM capabilities).

According to the RP2024 datasheet the UARTs also supoort break transmission (by setting bit 0 of UARTLCR_H).

So hopefully adding break support should only require a bit of plumbing in the debugprobe firmware and signalling and signalling break support in the CDC interface descriptor

P33M, 2024-03-18:
Can you give the associated PR branch a try?

HiassofT, 2024-03-18:
@P33M could you please upload / attach an uf2 file somewhere? I currently don't have a RP2024 dev env set up here

BTW: thanks a lot for the quick reply!

P33M, 2024-03-18:
[debugprobe.zip](https://github.com/raspberrypi/debugprobe/files/14636535/debugprobe.zip)

HiassofT, 2024-03-18:
Thanks a lot, a quick test with minicom looked fine and successfully triggered SYSRQ on my RPi5

Feel free to close this issue when the change is merged

closed 2024-03-18 by P33M