raspberrypi/debugprobe #132: Add version number to picotool output

jtraister, 2024-03-27:
Please add a bi_program_version_string() call with the version number. Right now I have to rely on release date to determine the difference between binaries, but that's not great.

lurch, 2024-03-27:
I guess this _sort of_ relates to #113 ?

jtraister, 2024-03-27:
I think so, just more info available about the binary.

fduxiao, 2024-03-28:
I'm not sure where to get the version string, but I wrote [this](https://github.com/fduxiao/picoprobe/commit/a94195f1756dd79643f7d1ccd41adb6570540b9e) adding the git tag as version. If HEAD is tagged, then the tag is used for version; otherwise, git commit hash is used. Is this reasonable?

What else is needed? I can add all these to binary information and make a pull request then.