raspberrypi/debugprobe #135: debugprobe not resonsive shortly after starting debug

MNS26, 2024-04-11:
whenever i try and use the picoprobe i end up with a debug console in VScode flooded with
```
Error: CMSIS-DAP command CMD_SWD_Configure failed.
Error: CMSIS-DAP failed to connect in mode (1)
Error: CMSIS-DAP command CMD_SWD_Configure failed.
Error: CMSIS-DAP failed to connect in mode (1)
Error: CMSIS-DAP command CMD_SWD_Configure failed.
Error: CMSIS-DAP failed to connect in mode (1)
Error: CMSIS-DAP command CMD_DISCONNECT failed.
Error: CMSIS-DAP failed to connect in mode (1)
```
and the ocational
```
Error: CMSIS-DAP failed to connect in mode (1)
Error: CMSIS-DAP transfer count mismatch: expected 3, got 1
Info : Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0
Error: CMSIS-DAP failed to connect in mode (1)
```
this used to not happen
whenever this happens i have to powercycle the debug probe to fix it
[debug console log (pastebin)](https://pastebin.com/UFPVtLGe)

i have no idea what could be causing this issue and it makes it reeeeally hard to debug projects

MNS26, 2024-04-12:
tried setting the adapter speed to 500KHz and still the same issue

the moment i try and get the memory location of the buffer it broke (hovering mouse over it)
[image]

lurch, 2024-04-12:
Ahh I see from your logfile that you're using `Open On-Chip Debugger 0.11.0-g8e3c38f78-dirty` so perhaps that's too old? https://github.com/raspberrypi/openocd is using `rp2040-v0.12.0`.

MNS26, 2024-04-12:
managed to trigger it again but this time it threw a error message as well 
that memory area is 255 bytes and where the I2C buffer lives 
(i had to manually re-add it to the memory tab since the code changed)

[image]

MNS26, 2024-04-12:
> Ahh I see from your logfile that you're using `Open On-Chip Debugger 0.11.0-g8e3c38f78-dirty` so perhaps that's too old? https://github.com/raspberrypi/openocd is using `rp2040-v0.12.0`.

hmm odd... i never manually installed any openocd version. i will try that later (irl stuff -_-)

MNS26, 2024-04-12:
hmm just had a look and i dont see a easy way to update the one from PlatformIO, the latest one they have is 10 months old [PlatformIO openocd](https://registry.platformio.org/tools/platformio/tool-openocd/versions)

MNS26, 2024-04-12:
nope. . . using their latest version manually specified
[image]

and the output of the [debug console](https://pastebin.com/LX1UYQhH)

MNS26, 2024-04-12:
oh interesting... even though i have VSCode closed it still has PlatformIO open.
I wonder if this might be a PlatformIO issue instead (i honestly dont know)
[image]

MNS26, 2024-04-13:
i remade my PlatformIO setup and now it is working (i quite dont know why but it works)

I will close the issue now

closed 2024-04-13 by MNS26