raspberrypi/debugprobe #141: Debugger starts only once on Windows

Slion, 2024-06-12:
Pico W + Debug Probe + Windows 11 + Pico Visual Studio from Pico Setup
The probe is updated to the latest firmware.

I can start the debugger once and it seems.to work fine. Then when I stop it the probe's yellow LED stays on.
Trying to restart it gives me OpenOCD timeout:
[image]

The only way to get it to work again is to reboot both Pico W and Probe.

lurch, 2024-06-13:
pinging @ndabas in case he has any suggestions

ndabas, 2024-06-13:
I'll have to try this out and see if I can replicate it. From the looks of it, it doesn't seem to be a Windows-specific issue ("probe's yellow LED stays on" and "Error: unable to find a matching CMSIS-DAP device".)

It's possible that something we're doing on Windows isn't closing the debugger connection cleanly though, so I'll take a look.

P33M, 2025-11-24:
Does this issue still exist with latest probe firmware and VScode extension?

jeffHODev, 2026-07-23:
#jeffho@CG20260723_114140(星期四)
#============================================================
最后按这个设定就解决了windows下面的调试连接问题了，在linux下面没有问题
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug (Cortex-Debug)",
            "cwd": "${userHome}/.pico-sdk/openocd/0.12.0+dev/scripts",
            "executable": "${command:raspberry-pi-pico.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "serverpath": "${userHome}/.pico-sdk/openocd/0.12.0+dev/openocd.exe",
            "gdbPath": "${command:raspberry-pi-pico.getGDBPath}",
            "debuggerArgs": [
                "-ex",
                "set debug-file-directory /debug"
            ],
            "device": "${command:raspberry-pi-pico.getChipUppercase}",
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "target/${command:raspberry-pi-pico.getTarget}.cfg"
            ],
            "svdFile": "${userHome}/.pico-sdk/sdk/2.2.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd",
            "runToEntryPoint": "main",
            // Fix for no_flash binaries, where monitor reset halt doesn't do what is expected
            // Also works fine for flash binaries
            // "overrideLaunchCommands": [
            //     "monitor reset init",
            //     "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            // ],
            // "openOCDLaunchCommands": [
            //     "adapter speed 5000"
            // ]
            // ==================== 修复部分 ====================
            "openOCDLaunchCommands": [
                "cmsis-dap backend hid", 
                "adapter speed 5000", 
                "reset_config none",
                "init",
                "reset init"
            ],
            "overrideLaunchCommands": [
                "monitor reset init",
                "monitor halt",
                "load"
            ],
        },
        {
            "name": "Pico Debug (Cortex-Debug with external OpenOCD)",
            "cwd": "${workspaceRoot}",
            "executable": "${command:raspberry-pi-pico.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "localhost:3333",
            "gdbPath": "${command:raspberry-pi-pico.getGDBPath}",
            "debuggerArgs": [
                "-ex",
                "set debug-file-directory /debug"
            ],
            "device": "${command:raspberry-pi-pico.getChipUppercase}",
            "svdFile": "${userHome}/.pico-sdk/sdk/2.2.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd",
            "runToEntryPoint": "main",
            // Fix for no_flash binaries, where monitor reset halt doesn't do what is expected
            // Also works fine for flash binaries
            "overrideLaunchCommands": [
                "monitor reset init",
                "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            ]
        },
    ]
}