raspberrypi/debugprobe #145: 2 files missing license header

recursivenomad, 2024-06-17:
The following files do not contain license information, and therefore cannot be forked/included in other projects:

- [probe_config.c](https://github.com/raspberrypi/debugprobe/blob/master/src/probe_config.c)
- [probe_oen.pio](https://github.com/raspberrypi/debugprobe/blob/master/src/probe_oen.pio)

As this has happened before (#5), it may be worth considering a license file in the root of the repository to serve as a fallback license for future files which lack license headers.

recursivenomad, 2024-08-14:
I have begun working on a derivative work and would like to use `probe_oen.pio` in it, however its lack of license header (and the project's lack of a root license file) means the code is "all rights reserved" and unable to legally be reused.

Although the file would be straightforward to recreate with clean-room design, I would prefer to be able to use and contribute back to the original file if possible.

P33M, 2024-11-13:
[FreeRTOS_Kernel_import.cmake](https://github.com/raspberrypi/debugprobe/blob/master/FreeRTOS_Kernel_import.cmake) and [pico_sdk_import.cmake](https://github.com/raspberrypi/debugprobe/blob/master/pico_sdk_import.cmake) are direct copies from the pico-sdk, which is BSD 3-clause, so I will add a header to those.
[probe_config.c](https://github.com/raspberrypi/debugprobe/blob/master/src/probe_config.c) and [probe_oen.pio](https://github.com/raspberrypi/debugprobe/blob/master/src/probe_oen.pio) are third-party contributions, and attribution/licence has been requested.
There are other contributors to [cmakelists.txt](https://github.com/raspberrypi/debugprobe/blob/master/CMakeLists.txt) and I willl CC them in the next comment.

All other files have a licence header - originally the project was designated MIT licenced, and going forward we'll specify that.

P33M, 2024-11-13:
@a-pushkin @newbrain @rewolff @fduxiao thank you for your contributions to this project, however we have a request.

Some of your edits were to the file [CMakeLists.txt](https://github.com/raspberrypi/debugprobe/blob/master/CMakeLists.txt) and this file was not tagged with an OSS licence (and neither was the overall project). Please confirm that you are the original authors of your work and that you are happy to contribute on the basis that this file is MIT licenced.

https://opensource.org/license/mit

rewolff, 2024-11-13:
I hereby declare that my contributions to this project can be included in this project and distributed and used under the licence that this project is licenced under, in this case the so-called MIT licence.

newbrain, 2024-11-20:
My contributions can be used and distributed according to the MIT license or any OSI registered form of the BSD license.
Please let me know if further action is needed on my part.

chrisboross, 2024-11-25:
@a-pushkin Sorry to nag! We'd really like to hear back from you?

P33M, 2024-11-27:
I have removed the remaining unattributed lines surviving in the project, asked @andygpz11 to do a cleanroom reimplemantation of the missing functionality, and have tagged it as MIT licensed.
https://github.com/raspberrypi/debugprobe/commit/65b1e73589f7248a6208b0bdcc32c8b8e7e169ee

closed 2025-01-09 by P33M