raspberrypi/debugprobe #148: USB Power Descriptor a little on the low side for -DDEBUG_ON_PICO=1

peardox, 2024-08-22:
In [usb_descriptors.c](https://github.com/raspberrypi/debugprobe/blob/master/src/usb_descriptors.c#L102C3-L102C24) the max power in ma is defined as 100ma via TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, 0, 100)

While this seems fine for the pre-built probe it does seem a little stingy if -DDEBUG_ON_PICO=1 as if for example you had a pico_w attached directly to a pico and sharing it's power it is quite easy to envisage transient larger power requirements

If a pico is switched to BOOTSEL then the USB descriptor rates it at 500ma - which seems a far more sensible value to use

lurch, 2024-08-22:
I was wondering if powering 2 Picos from a single USB port is a recommended option, but then I found it explicitly mentioned in the getting-started-guide:
[image]

P33M, 2024-11-07:
It's explicitly set low because otherwise and if plugged into a bus-powered hub, it will usually get ignored by the host OS. That's more irritating to debug for users.

closed 2024-11-07 by P33M